apache/mynewt-core #795: nrf52 thingy target ble stack does not come back up after an erase command

jacobrosenthal, 2018-02-06:
This is apparently not related to https://github.com/apache/mynewt-core/pull/789

```
$ newt target show split-thingy
targets/split-thingy
    app=@apache-mynewt-core/apps/blesplit
    bsp=@apache-mynewt-core/hw/bsp/nrf52-thingy
    build_profile=optimized
    loader=@apache-mynewt-core/apps/bleprph
    syscfg=CONSOLE_ECHO=0:CONSOLE_INPUT=0:CONSOLE_MAX_INPUT_LEN=0:CONSOLE_RTT=1:CONSOLE_UART=0:JLINK_DEBUG=1:LOG_LEVEL=0:SHELL_TASK=0
```

```
$ newt build split-thingy && newt create-image split-thingy 0.0.0.1 && newt load split-thingy 0.0.0.1
Building target targets/split-thingy
...
Loading app image into slot 2
Loading loader image into slot 1
$ newt create-image split-thingy 0.0.0.2
...
App image succesfully generated: /Users/jacobrosenthal/Downloads/chippd3/bin/targets/split-thingy/loader/apps/bleprph/bleprph.img
App image succesfully generated: /Users/jacobrosenthal/Downloads/chippd3/bin/targets/split-thingy/app/apps/blesplit/blesplit.img
$ ./test2.sh

**** Running: newtmgr -c nimble_bleprph image erase *****

Done

**** Running: newtmgr -c nimble_bleprph image list *****

Images:
 slot=0
    version: 0.0.0.1
    bootable: true
    flags: active confirmed
    hash: a5cc4e507442d75e130f9f2ae69b569f21eb9da3e74c50a57d6a4fded31c99e0
Split status: N/A (0)

**** Running: newtmgr -c nimble_bleprph image upload /Users/jacobrosenthal/Downloads/chippd3/bin/targets/split-thingy/loader/apps/bleprph/bleprph.img *****

 125.98 KiB / 126.16 KiB [==============================================================================================================]  99.85%Done

**** Running: newtmgr -c nimble_bleprph image list *****

Images:
 slot=0
    version: 0.0.0.1
    bootable: true
    flags: active confirmed
    hash: a5cc4e507442d75e130f9f2ae69b569f21eb9da3e74c50a57d6a4fded31c99e0
 slot=1
    version: 0.0.0.2
    bootable: true
    flags: 
    hash: 517d45dace367a948a771e329f9efd890ef4d367f11b3a44b5104aa5455c8a56
Split status: N/A (0)

**** Running: newtmgr -c nimble_bleprph image erase *****

Error: disconnected

**** Running: newtmgr -c nimble_bleprph image list *****

Error: Failed to connect to peer after 10s
```

logs
```
000001 [ts=7812ssb, mod=64 level=0] registered service 0x1800 with handle=1
000002 [ts=15624ssb, mod=64 level=0] registering characteristic 0x2a00 with def_handle=2 val_handle=3
[... 158 lines cut ...]
008052 [ts=62906192ssb, mod=4 level=0] ble_hs_hci_evt_acl_process(): conn_handle=1 pb=1 len=27 data=0x61 0x58 0x50 0x3c 0xb8 0xf3 0x96 0x24 0x00 0x00 0x00 0x20 0x00 0x00 0x00 0x64 0xf8 0x01 0x00 0x02 0x00 0x00 0x00 0x00 0x00 0x00 0x00 
008053 [ts=62914004ssb, mod=4 level=0] ble_hs_hci_evt_acl_process(): conn_handle=1 pb=1 len=27 data=0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x20 0xb5 0x82 0x00 0x00 0x11 0x83 0x00 0x00 0x13 0x83 0x00 0x00 0x00 0x00 0x00 
00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x5b 0x94 
008054 [ts=62921816ssb, mod=4 level=0] rxed att command: prep write req; conn=1 handle=0x001f offset=0
008054 [ts=62921816ssb, mod=4 level=0] txed att command: prep write rsp; conn=1 handle=0x001f offset=0
008054 [ts=62921816ssb, mod=4 level=0] host tx hci data; handle=1 length=108
008054 [ts=62921816ssb, mod=4 level=0] ble_hs_hci_acl_tx(): 0x01 0x00 0x6c 0x00 0x68 0x00 0x04 0x00 0x17 0x1f 0x00 0x00 0x00 0x02 0x00 0x00 0x66 0x00 0x01 0x01 0x01 0xa3 0x63 0x6f 0x66 0x66 0x00 0x64 0x64 0x61 0x74 0x61 0x58 0x50 0x3c 0xb8 0xf3 0x96 0x24 0x00 0x00 0x00 0x20 0x00 0x00 0x00 0x64 0xf8 0x01 0x00 0x02 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x01 0x20 0xb5 0x82 0x00 0x00 0x11 0x83 0x00 0x00 0x13 0x83 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x00 0x5b 0x94 
008060 [ts=62968688ssb, mod=4 level=0] ble_hs_hci_evt_acl_process(): conn_handle=1 pb=2 len=20 data=0x10 0x00 0x04 0x00 0x16 0x1f 0x00 0x63 0x00 0x00 0x00 0x63 0x6c 0x65 0x6e 0x1a 0x00 0x01 0xf8 0xa8 
008060 [ts=62968688ssb, mod=4 level=0] rxed att command: prep write req; conn=1 handle=0x001f offset=99
008060 [ts=62968688ssb, mod=4 level=0] txed att command: prep write rsp; conn=1 handle=0x001f offset=99
008060 [ts=62968688ssb, mod=4 level=0] host tx hci data; handle=1 length=20
008060 [ts=62968688ssb, mod=4 level=0] ble_hs_hci_acl_tx(): 0x01 0x00 0x14 0x00 0x10 0x00 0x04 0x00 0x17 0x1f 0x00 0x63 0x00 0x00 0x00 0x63 0x6c 0x65 0x6e 0x1a 0x00 0x01 0xf8 0xa8 
008060 [ts=62968688ssb, mod=4 level=0] Number of Completed Packets: num_handles=1
008060 [ts=62968688ssb, mod=4 level=0] handle:1 pkts:1
008063 [ts=62992124ssb, mod=4 level=0] ble_hs_hci_evt_acl_process(): conn_handle=1 pb=2 len=6 data=0x02 0x00 0x04 0x00 0x18 0x01 
008064 [ts=63000000ssb, mod=4 level=0] rxed att command: exec write req; conn=1 flags=0x01
008064 [ts=63000000ssb, mod=4 level=0] host tx hci data; handle=1 length=5
008064 [ts=63000000ssb, mod=4 level=0] ble_hs_hci_acl_tx(): 0x01 0x00 0x05 0x00 0x01 0x00 0x04 0x00 0x19 
008064 [ts=63000000ssb, mod=4 level=0] Number of Completed Packets: num_handles=1
008064 [ts=63000000ssb, mod=4 level=0] handle:1 pkts:1
008079 [ts=63117180ssb, mod=4 level=1] GATT procedure initiated: notify; att_handle=31
008080 [ts=63124992ssb, mod=4 level=0] txed att command: notify req; conn=1 handle=0x001f
008080 [ts=63124992ssb, mod=4 level=0] host tx hci data; handle=1 length=27
008080 [ts=63124992ssb, mod=4 level=0] ble_hs_hci_acl_tx(): 0x01 0x00 0x1b 0x00 0x17 0x00 0x04 0x00 0x1b 0x1f 0x00 0x03 0x00 0x00 0x0c 0x00 0x01 0x01 0x01 0xbf 0x62 0x72 0x63 0x00 0x63 0x6f 0x66 0x66 0x18 0x50 0xff 
```

andrzej-kaczmarek, 2018-02-13:
I think this is the same issue as #796 so please check fix for that issue.

andrzej-kaczmarek, 2018-02-14:
Fixed by #810

closed 2018-02-14 by andrzej-kaczmarek