apache/mynewt-core #801: Split status: non-matching (1) when (near?) out of memory

jacobrosenthal, 2018-02-09:
I keep randomly running into this and just figurd it out. When split targets are low memory on one or the other image, they can show as Split status: non-matching (1) after you test into the app

```
$ newt target show split-microbit
targets/split-microbit
    app=@apache-mynewt-core/apps/blesplit
    bsp=@apache-mynewt-core/hw/bsp/bbc_microbit
    build_profile=optimized
    loader=@apache-mynewt-core/apps/bleprph
    syscfg=BLE_LL_CFG_FEAT_LE_ENCRYPTION=0:BLE_SM_LEGACY=0:BLE_SM_SC=0:LOG_LEVEL=255:MSYS_1_BLOCK_COUNT=19
```

```
$ newtmgr -cnimble_blesplit image list
Images:
 slot=0
    version: 0.0.0.1
    bootable: true
    flags: active confirmed
    hash: dced80ca9c6c18500c8035d31afb3d41c900e4de7defeaa1c27c083f13125d7c
 slot=1
    version: 0.0.0.1
    bootable: false
    flags: active confirmed
    hash: d04aae5bf5bc8ae495e1d59b6211600a9de983ab851be2cbe6d9c759b86d68d7
Split status: non-matching (1)
```

Clearly it 'runs' enough to respond to an image list command, but it also doesnt think its matchin. But if I decrease block count and give some more breathing room:

```
$ newt target show split-microbit
targets/split-microbit
    app=@apache-mynewt-core/apps/blesplit
    bsp=@apache-mynewt-core/hw/bsp/bbc_microbit
    build_profile=optimized
    loader=@apache-mynewt-core/apps/bleprph
    syscfg=BLE_LL_CFG_FEAT_LE_ENCRYPTION=0:BLE_SM_LEGACY=0:BLE_SM_SC=0:LOG_LEVEL=255:MSYS_1_BLOCK_COUNT=18
```

```
$ newtmgr -cnimble_blesplit image list
Images:
 slot=0
    version: 0.0.0.1
    bootable: true
    flags: active confirmed
    hash: 9ed01981c27d6aa45003a322ec4091ebcc62173a1a84e79bdbc803e5c1d9a219
 slot=1
    version: 0.0.0.1
    bootable: false
    flags: active
    hash: 3e74b1cd8f2e76629160cfd6f91e9ce6726f4ca289f9c8707718873911a5d148
Split status: matching (2)
```

Maybe a feature not a bug? Documenting here either way.

ccollins476ad, 2018-02-09:
By "memory", do you mean stack size?  A stack overflow will yield unpredictable behavior, and I'm afraid not much can be done about that.

jacobrosenthal, 2018-02-09:
No idea. All I know is what I posted above, so in future if anyone is
seeing non-matching (1), it might be a memory issue, stack or otherwise. I
Don't really care if this is marked wont fix, just wanted to document it as
Ive been randomly seeing it over the last month and couldnt figure out if
it was a specific target or newt or newtmgr bug, and now I believe it has
something to do with newt-core and low memory.

closed 2018-02-19 by jacobrosenthal