apache/mynewt-core #805: Mac OS High Sierra DFU Error

adrien3d, 2018-02-12:
I am trying to follow this [tutorial](https://mynewt.apache.org/latest/os/tutorials/ota_upgrade_nrf52/), but I have a bug when trying to do 
`newtmgr image list -c mybleprph`

It returns me:
`2018/02/12 10:31:16 Unhandled event: xpc.Dict{"kCBMsgId":4, "kCBMsgArgs":xpc.Dict{"kCBMsgArgState":5}}`
And then nothing.

Do you have any idea of where the problem comes from?
I think it is related to the fact that I am on Mac OS High Sierra.

I have found that you are using paypall/gatt like [library](https://github.com/runtimeco/gatt) for DFU over BLE with MacOS/Linux.
While googling the error, I discovered that it is an error that go-ble/ble, and moogle19/ble have fixed [here](https://github.com/go-ble/ble/pull/2) and partially [here](https://github.com/moogle19/ble/commit/2bf67be0f6a2e7d65f6e5a4f3f15b0ea31af2ad3) 

Do you know how I can change the library used by newtmgr to connect over ble, or where the sources used by newtmgr library are located?

mkiiskila, 2018-02-12:
Hi Adrian,

fix should’ve been merged in here: https://github.com/apache/mynewt-newtmgr/pull/53 <https://github.com/apache/mynewt-newtmgr/pull/53>

adrien3d, 2018-02-12:
On mac OS High Sierra, while doing a `newtmgr image upload -c bleconn /path/to/my/bleprph.img`, I have a progress bar remaining at 0B, and an Error: disconnected:
`newtmgr image upload -c bleconn /path/to/my/bleprph.img
 0 B / 122.70 KiB [------------------------------------------------------------------------------------------------------------------------------------------------------]   0.00%
Error: disconnected
`

Any idea of where the problem is coming from?

jacobrosenthal, 2018-02-12:
Thanks for reporting. Were tracking this here already https://github.com/apache/mynewt-core/issues/796

adrien3d, 2018-02-12:
Not so sure of this, because `newtmgr -c bleconn image list` returns me nothings, even after loading the bleprph example:
`Images:
Split status: N/A (0)`

jacobrosenthal, 2018-02-12:
Please give way more logs, I have no idea where in the process image list is failing for you. I find most failures like what youve described above are an out of memory or overflow of some type, which means youll need to fully erase and load everything again after such an error. This is what my loading script looks like for my nrf52 split-thingy target, rename for yours:
```
echo -ne "erase all\nq\n" | JLinkExe -if SWD -device nRF52 -speed 1000
newt build boot-thingy && newt load boot-thingy
newt build split-thingy && newt create-image split-thingy 0.0.0.1 && newt load split-thingy 0.0.0.1
```

adrien3d, 2018-02-12:
Okay, thanks, I'll give this a try.
I have currently: `newtmgr -c bleconn image list -l debug`

`DEBU[2018-02-12 17:17:52.188] Using connection profile: name=bleconn type=ble connstring=peer_name=klocz-ble 
DEBU[2018-02-12 17:17:52.2] {add-oic-listener} [transceiver.go:62] token=[] 
DEBU[2018-02-12 17:17:52.2] Connecting to peer                           
DEBU[2018-02-12 17:17:52.597] Exchanging MTU                               
DEBU[2018-02-12 17:17:52.597] Exchanged MTU; ATT MTU = 104                 
DEBU[2018-02-12 17:17:52.597] Discovering profile                          
DEBU[2018-02-12 17:17:53.339] Subscribing to NMP response characteristic   
DEBU[2018-02-12 17:17:53.384] {add-nmp-listener} [bll_sesn.go:401] seq=66  
DEBU[2018-02-12 17:17:53.384] Encoded &{NmpBase:{hdr:{Op:0 Flags:0 Len:0 Group:1 Seq:66 Id:0}}} to:
00000000  a0                                                |.|
DEBU[2018-02-12 17:17:53.384] Encoded:
00000000  00 00 00 01 00 01 42 00  a0                       |......B..|
DEBU[2018-02-12 17:17:53.384] Tx NMP request: 00000000  00 00 00 01 00 01 42 00  a0                       |......B..|
DEBU[2018-02-12 17:17:53.431] Received nmp rsp: &{NmpBase:{hdr:{Op:1 Flags:0 Len:24 Group:1 Seq:66 Id:0}} Rc:0 Images:[] SplitStatus:N/A} 
DEBU[2018-02-12 17:17:53.431] {remove-nmp-listener} [bll_sesn.go:401] seq=66 
Images:
Split status: N/A (0)`

jacobrosenthal, 2018-02-12:
Also check it out, if you use triple ticks for large code blocks it will maintain your line breaks and format it properly.

adrien3d, 2018-02-13:
I also have logs on the nRF52-DK side:

For image list:
```
000323 [ts=2523404ssb, mod=64 level=1] connection established; status=0 handle=1 our_ota_addr_type=0 our_ota_addr=00:01:02:03:04:05 our_id_addr_type=0 our_id_addr=00:01:02:03:04:05 peer_ota_addr_type=0 peer_ota_addr=6c:40:08:b2:82:1d peer_id_addr_type=0 peer_id_addr=my_mac_addr conn_itvl=9 conn_latency=0 supervision_timeout=200 encrypted=0 authenticated=0 bonded=0
000332 [ts=2593712ssb, mod=64 level=1] 
000333 [ts=2601524ssb, mod=64 level=1] mtu update event; conn_handle=1 cid=4 mtu=104
000363 [ts=2835884ssb, mod=64 level=1] subscribe event; conn_handle=1 attr_handle=14 reason=1 prevn=0 curn=0 previ=0 curi=1
000388 [ts=3031248ssb, mod=64 level=1] subscribe event; conn_handle=1 attr_handle=31 reason=1 prevn=0 curn=1 previ=0 curi=0
000393 [ts=3070308ssb, mod=4 level=1] GATT procedure initiated: notify; att_handle=31
000396 [ts=3093744ssb, mod=64 level=1] connection updated; status=531 handle=1 our_ota_addr_type=0 our_ota_addr=00:01:02:03:04:05 our_id_addr_type=0 our_id_addr=00:01:02:03:04:05 peer_ota_addr_type=0 peer_ota_addr=my_mac_addr peer_id_addr_type=0 peer_id_addr=my_mac_addr conn_itvl=9 conn_latency=0 supervision_timeout=200 encrypted=0 authenticated=0 bonded=0
000405 [ts=3164052ssb, mod=64 level=1] 
000406 [ts=3171864ssb, mod=64 level=1] subscribe event; conn_handle=1 attr_handle=14 reason=2 prevn=0 curn=0 previ=1 curi=0
000409 [ts=3195300ssb, mod=64 level=1] subscribe event; conn_handle=1 attr_handle=31 reason=2 prevn=1 curn=0 previ=0 curi=0
000413 [ts=3226548ssb, mod=64 level=1] disconnect; reason=531 handle=1 our_ota_addr_type=0 our_ota_addr=00:01:02:03:04:05 our_id_addr_type=0 our_id_addr=00:01:02:03:04:05 peer_ota_addr_type=0 peer_ota_addr=my_mac_addr peer_id_addr_type=0 peer_id_addr=my_mac_addr conn_itvl=9 conn_latency=0 supervision_timeout=200 encrypted=0 authenticated=0 bonded=0
000422 [ts=3296856ssb, mod=64 level=1] 
000423 [ts=3304668ssb, mod=4 level=1] GAP procedure initiated: advertise; disc_mode=2 adv_channel_map=0 own_addr_type=0 adv_filter_policy=0 adv_itvl_min=0 adv_itvl_max=0
```

For image upload:
```
002808 [ts=21937440ssb, mod=64 level=1] connection established; status=0 handle=1 our_ota_addr_type=0 our_ota_addr=00:01:02:03:04:05 our_id_addr_type=0 our_id_addr=00:01:02:03:04:05 peer_ota_addr_type=0 peer_ota_addr=my_mac_addr peer_id_addr_type=0 peer_id_addr=my_mac_addr conn_itvl=9 conn_latency=0 supervision_timeout=200 encrypted=0 authenticated=0 bonded=0
002818 [ts=22015624ssb, mod=64 level=1] 
002819 [ts=22023436ssb, mod=64 level=1] mtu update event; conn_handle=1 cid=4 mtu=104
002850 [ts=22265608ssb, mod=64 level=1] subscribe event; conn_handle=1 attr_handle=14 reason=1 prevn=0 curn=0 previ=0 curi=1
002874 [ts=22453096ssb, mod=64 level=1] subscribe event; conn_handle=1 attr_handle=31 reason=1 prevn=0 curn=1 previ=0 curi=0
```

adrien3d, 2018-02-14:
And with hci dump: 
For listing images
```
HCI sniffer - Bluetooth packet analyzer ver 5.43
device: hci0 snap_len: 1500 filter: 0xffffffffffffffff
2018-02-14 12:07:06.050942 > HCI Event: Command Complete (0x0e) plen 4
    Reset (0x03|0x0003) ncmd 1
    status 0x00
2018-02-14 12:07:06.054091 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Supported Features (0x04|0x0003) ncmd 1
    status 0x00
    Features: 0xff 0xff 0x8f 0xfe 0xdb 0xff 0x5b 0x87
2018-02-14 12:07:06.056962 > HCI Event: Command Complete (0x0e) plen 12
    Read Local Version Information (0x04|0x0001) ncmd 1
    status 0x00
    HCI Version: 4.0 (0x6) HCI Revision: 0x22bb
    LMP Version: 4.0 (0x6) LMP Subversion: 0x22bb
    Manufacturer: Cambridge Silicon Radio (10)
2018-02-14 12:07:06.059951 > HCI Event: Command Complete (0x0e) plen 10
    Read BD ADDR (0x04|0x0009) ncmd 1
    status 0x00 bdaddr 00:1A:7D:DA:71:0A
2018-02-14 12:07:06.062956 > HCI Event: Command Complete (0x0e) plen 11
    Read Buffer Size (0x04|0x0005) ncmd 1
    status 0x00
    ACL MTU 310:10 SCO MTU 64:8
2018-02-14 12:07:06.065960 > HCI Event: Command Complete (0x0e) plen 7
    Read Class of Device (0x03|0x0023) ncmd 1
    status 0x00 class 0x000000
2018-02-14 12:07:06.104984 > HCI Event: Command Complete (0x0e) plen 252
    Read Local Name (0x03|0x0014) ncmd 1
    status 0x00 name 'CSR8510 A10'
2018-02-14 12:07:06.108079 > HCI Event: Command Complete (0x0e) plen 6
    Read Voice Setting (0x03|0x0025) ncmd 1
    status 0x00 voice setting 0x0060
2018-02-14 12:07:06.112979 > HCI Event: Command Complete (0x0e) plen 5
    Read Number of Supported IAC (0x03|0x0038) ncmd 1
2018-02-14 12:07:06.115945 > HCI Event: Command Complete (0x0e) plen 8
    Read Current IAC LAP (0x03|0x0039) ncmd 1
    IAC 0x9e8b33 (General Inquiry Access Code)
2018-02-14 12:07:06.119031 > HCI Event: Command Complete (0x0e) plen 4
    Set Event Filter (0x03|0x0005) ncmd 1
    status 0x00
2018-02-14 12:07:06.122156 > HCI Event: Command Complete (0x0e) plen 4
    Write Connection Accept Timeout (0x03|0x0016) ncmd 1
    status 0x00
2018-02-14 12:07:06.125935 > HCI Event: Command Complete (0x0e) plen 7
    LE Read Buffer Size (0x08|0x0002) ncmd 1
    status 0x00 pktlen 0x0000 maxpkt 0x00
2018-02-14 12:07:06.128994 > HCI Event: Command Complete (0x0e) plen 12
    LE Read Local Supported Features (0x08|0x0003) ncmd 1
    status 0x00 features 0x0100000000000000 (Link Layer supports LE Encryption)
2018-02-14 12:07:06.131953 > HCI Event: Command Complete (0x0e) plen 12
    LE Read Supported States (0x08|0x001c) ncmd 1
2018-02-14 12:07:06.144070 > HCI Event: Command Complete (0x0e) plen 68
    Read Local Supported Commands (0x04|0x0002) ncmd 1
    status 0x00
    Commands: ffffff03fefffffffffffffff30fe8fe3ff783ff1c00000061f7ffff7f
2018-02-14 12:07:06.147020 > HCI Event: Command Complete (0x0e) plen 4
    Write Simple Pairing Mode (0x03|0x0056) ncmd 1
    status 0x00
2018-02-14 12:07:06.150933 > HCI Event: Command Complete (0x0e) plen 4
    Write Inquiry Mode (0x03|0x0045) ncmd 1
    status 0x00
2018-02-14 12:07:06.154168 > HCI Event: Command Complete (0x0e) plen 5
    Read Inquiry Response Transmit Power Level (0x03|0x0058) ncmd 1
    status 0x00 level 4
2018-02-14 12:07:06.156969 > HCI Event: Command Complete (0x0e) plen 14
    Read Local Extended Features (0x04|0x0004) ncmd 1
    status 0x00 page 1 max 1
    Features: 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00
[... 462 lines cut ...]
2018-02-14 12:11:01.422594 > HCI Event: Command Complete (0x0e) plen 6
    Delete Stored Link Key (0x03|0x0012) ncmd 1
    status 0x00 deleted 0
2018-02-14 12:11:01.425555 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Advertising Data (0x08|0x0008) ncmd 1
    status 0x00
2018-02-14 12:11:01.428447 > HCI Event: Command Complete (0x0e) plen 4
    LE Set Scan Response Data (0x08|0x0009) ncmd 1
    status 0x00
2018-02-14 12:11:01.431366 > HCI Event: Command Complete (0x0e) plen 4
    Write Class of Device (0x03|0x0024) ncmd 1
    status 0x00
2018-02-14 12:11:01.434524 > HCI Event: Command Complete (0x0e) plen 4
    Write Local Name (0x03|0x0013) ncmd 1
    status 0x00
2018-02-14 12:11:01.438306 > HCI Event: Command Complete (0x0e) plen 4
    Write Extended Inquiry Response (0x03|0x0052) ncmd 1
    status 0x00
2018-02-14 12:11:01.444641 > HCI Event: Command Complete (0x0e) plen 4
    Write Extended Inquiry Response (0x03|0x0052) ncmd 1
    status 0x00
2018-02-14 12:11:01.449279 > HCI Event: Command Complete (0x0e) plen 4
    Write Extended Inquiry Response (0x03|0x0052) ncmd 1
    status 0x00
```

adrien3d, 2018-02-14:
@jacobrosenthal I switch my `repository.apache-mynewt-core to vers: 1.1.0` in `project.yml`, and now the OTA DFU between and embedded Broadcomm Macbook Pro 2015 is working.
I am investigating to see what commit since the 1.1.0 breaks the DFU.

adrien3d, 2018-02-14:
@jacobrosenthal I updated newt and newtmgr to Head and now the OTA DFU between and embedded Broadcomm Macbook Pro 2015 is working.

closed 2018-02-14 by adrien3d

jacobrosenthal, 2018-02-14:
Happy to hear that. Keep in touch check out the slack too at
mynewt.slack.com

adrien3d, 2018-02-15:
Can you invite me? adrien3d[at]gmail[dot]com

mkiiskila, 2018-02-15:
Follow link at http://mynewt.apache.org bottom of the page.