apache/mynewt-core #826: soft pwm api questions

jacobrosenthal, 2018-02-21:
Current softpwm uses os cpu time, ie I think that means it shares with the system timer. At how often a timer would that get problematic?

Would it be better to use a separate hal timer? or I guess maybe a syscfg flag to choose the source?

Secondly,
soft pwm appears to attempt to utilize the pwm api 1:1, why should it get a separate device name?
https://github.com/apache/mynewt-core/blob/master/apps/pwm_test/src/main.c#L41

I would argue for the default bsps to resolve soft pwm to pwm0

mlaz, 2018-02-21:
> I would argue for the default bsps to resolve soft pwm to pwm0

It made sense at the time having a name which tells us it is not the hardware based PWM, plus using it won't affect any other PWM device, which might happen with hw based PWM. Is this against any pattern we're following with device names? Probably, so if everyone agrees let's change it.

wes3, 2018-02-21:
I am not saying having a syscfg (or not having one) is a good/bad idea, but I would think it would be fine in most cases. Hard to say without getting hard numbers but it is not something I would be particularly worried about. What period/frequencies are being generally used here?

closed 2018-03-21 by jacobrosenthal