apache/mynewt-core #864: JSON library's array count not incrementing

adityaxavier, 2018-03-01:
Firstly, there is almost no documentation regarding json utilization.

After following the example and documentation of mJSON which is not entirely the same as mynewt's JSON library. I was able to get the code working partially.

Which this code, am getting the following output

`000001 Buffer Initiated 
000002 JSON Read 0! 
000003 PRN = 10, elevation = 45, azimuth = 196 
Complete
`
[blinky.zip](https://github.com/apache/mynewt-core/files/1771077/blinky.zip)

michal-narajowski, 2018-03-07:
Hi @adityaxavier,

I found some issues with your code and an issue in JSON library code. I submitted a pull request here: https://github.com/apache/mynewt-core/pull/884

The updated app code that should work fine can be found on my branch: 
https://github.com/michal-narajowski/incubator-mynewt-core/tree/json-fix-app

Best regards,
Michał Narajowski

michal-narajowski, 2018-03-21:
@adityaxavier, does this solve your issue?

adityaxavier, 2018-03-21:
@michal-narajowski, Yes it does..

closed 2018-03-21 by adityaxavier