apache/mynewt-core #876: UART logging caused an exception for nrf51 board

sterlinghughes, 2018-03-06:
Blinking code work well, but when started to play with nimble BLE stack I got an issue. So problem is - even with bare minimum BLE code from tutorial section my board (ble400) does not work - I got only TX led blinking. 

So expecting some debug output on UART plugged in USB->UART moduel to it and get this in terminal

 

000000 Assert @ 0xa8cb␍␊
000000 Unhandled interrupt (2), exception sp 0x20001168␍␊
000000 r0:0x00000000 r1:0x00000000 r2:0x80000000 r3:0xe000ed00␍␊
000000 r4:0x0000a8cb r5:0x00000002 r6:0xffffffff r7:0x000091a5␍␊
000000 r8:0xffffffff r9:0xffffffff r10:0xffffffff r11:0xffffffff␍␊
000000 r12:0x00000001 lr:0x0000925f pc:0x000081cc psr:0x81000000␍␊
000000 ICSR:0x00421002␍␊
This block repeating. 

Finally I got the reason caused it

If specify in pkg.yml

pkg.deps: 
...
  - "@apache-mynewt-core/sys/log/full" 
  - "@apache-mynewt-core/sys/stats/full"
...
then this issue appear. If replace full to stub - everything work well. 

So obviously something is wrong with those 2 packages when running on nrf51 MCU. And really logging to UART is working - just causing some exception and corresponding application crash after data transmitted.  

BTW as I discovered interrupt 2 is an UART interrupt for nrf51 SDK.

michal-narajowski, 2018-04-11:
I investigated this issue and it seems that there is not enough RAM to run this app with default settings on ble400 with 16kB of RAM. After you replaced logs and stats packages with stubs you reduced RAM usage and the app could successfully run.

utzig, 2018-04-11:
I agree that RAM is the reason, but if it was due to static allocation the linker would catch it, if it is `malloc` someone is not testing the return value, or it might have a bug. What do you think?

michal-narajowski, 2018-04-11:
This error happens during sysinit in ble_hci_ram_pkg_init() so it is `malloc`.

andrzej-kaczmarek, 2018-04-11:
so transport packages should use static buffers for mempools - there's no point of mallocing them if we know size at compile time and failing to malloc buffer results in an assert anyway. the same was done e.g. for btshell in https://github.com/apache/mynewt-core/pull/812

utzig, 2018-04-12:
so it is caused by allocation error followed by `SYSINIT_PANIC_ASSERT(rc == 0)`?

michal-narajowski, 2018-04-13:
Exactly. Here is a PR that replaces malloc with static allocation in nimble transport: https://github.com/apache/mynewt-nimble/pull/40

mlaz, 2018-05-17:
https://github.com/apache/mynewt-nimble/pull/40 is merged

closed 2018-05-17 by mlaz