apache/mynewt-core #895: Nimble host tests are not running due to compilation issue

mehulhirpara, 2018-03-09:
I have create new project to work on NimBLE and trying to run tests but found compilation issues. 

Build logs for reference: [nimble_host_test_compilation_log.txt](https://github.com/apache/mynewt-core/files/1796624/nimble_host_test_compilation_log.txt)

Working solution is to initialize structure in following way:
struct ble_hs_test_util_hci_num_completed_pkts_entry ncpe[2] = **{ {0} }**;

instead of:
struct ble_hs_test_util_hci_num_completed_pkts_entry ncpe[2] = **{ 0 }**;

ccollins476ad, 2018-03-09:
Thanks for reporting this!

`{ 0 }` is valid C syntax for default initializing a struct.  This is an instance of my most dreaded gcc bug: https://gcc.gnu.org/bugzilla/show_bug.cgi?id=53119

This gcc bug was fixed in 2014, but the bug lives on in older versions of gcc.

I suggest upgrading to a newer version of gcc.  We should also apply your fix to support older versions of gcc.

sjanc, 2018-05-16:
Those should be fixed now by using memset() instead. If you stil have any issues with compiling those please open separate issue against mynewt-nimble repo.

closed 2018-05-16 by sjanc