apache/mynewt-core #955: pwm enabled?

sdalu, 2018-03-23:
It's seems that it is only permitted to call pwm_disable if it was already pwm_enable_* before (tested on nrf52).
An API for testing if the pwm is currently enabled would be useful.
The same goes for configured state, as pwm_disable unconfigure the channel

Perhaps a diagram showing what function can be called in what order would be a nice addition, as well as explicitly saying in the documentation when it its legit to call the function

mlaz, 2018-03-23:
This seems reasonable to me. On it.

mlaz, 2018-03-28:
> An API for testing if the pwm is currently enabled would be useful.

I'm PRing this soon along with a README for pwm_test.

> The same goes for configured state, as pwm_disable unconfigure the channel

@sdalu What you mean by this? PWM implementations usually get some default values when you don't configure them. If you don't assign a pin to a channel it doesn't make much sense to do anything with it. I don't see a case when someone needs to query whether a channel has been configured.

> Perhaps a diagram showing what function can be called in what order would be a nice addition, as well as explicitly saying in the documentation when it its legit to call the function

A README file for apps/pwm_test is in the making but I think it wouldn't be hard to agree that you need to open a device, then configure a channel, then set a duty cycle on it (i.e. enable). So, I'll make sure the PWM API documentation is updated with the requirements for calling each function (i.e. calling pwm_enable_duty_cycle requiring the channel to be already configured) and that this somehow is mentioned on the README.

sdalu, 2018-03-29:
I would naturally use this sequence:  pwm_chan_config, pwm_enable_duty_cycle, pwm_disable, pwm_enable_duty_cycle, pwm_disable, ...

But pwm_disable documentation says : Disable the PWM channel, it will be marked as unconfigured.
So it seems the less obvious sequence will be:  pwm_chan_config, pwm_enable_duty_cycle, pwm_disable, pwm_chan_config, pwm_enable_duty_cycle, pwm_disable

Also are all the pwm_* functions allowed to be run in interrupt context? (cycle_handler, seq_end_handler)?

What happen when:
 * cycles are finished: is the pwm just stopped, is it disabled, unconfigured ?
 * a call to pwm_enable is made, what are the consequences on the cycle counter? (in normal thread, in seq_end_handler, in cycle_handler)

Is it possible to call pwm_set_frequency, after pwm_chan_config, pwm_enable?
(what will be the consquences?)

Is it possible to know the cycle couter value in pwm_cycle_handler?

mlaz, 2018-04-13:
Ok, I see 2 issues here, probably we want :
1 - pwm_disable to just make the channel stop and not unconfigure it.
2 - Closing the device will make every channel unconfigured.
3 - pwm_is_enabled because we now can make a channel run for a number of cycles and may want to know if it has already been configured.
4 - A way to make a channel unassigned to a pin. (I'll make a PWM_NO_PIN macro so the user may do this via configure_channel)

mlaz, 2018-04-13:
> What happen when:
>
> cycles are finished: is the pwm just stopped, is it disabled, unconfigured ?

Should be configured but disabled.

> a call to pwm_enable is made, what are the consequences on the cycle counter? (in normal thread, in seq_end_handler, in cycle_handler)

If seq_end_handler is set it should be fired whenever the PWM stops (regardless it is playing for a given number of cycles or in loop (i.e. n_cycles = 0)).
If cycle_handler is set it is fired every cycle (so, yes be careful when using it on high freq).

> Is it possible to call pwm_set_frequency, after pwm_chan_config, pwm_enable?
> (what will be the consquences?)

It should be, the PWM should restart automatically on the new freq.

> Is it possible to know the cycle couter value in pwm_cycle_handler?

No (not right now), you would have to use am external variable. This sounds like an interesting improvement although would compromise the generality of the handler [prototype](https://github.com/apache/mynewt-core/blob/8ac604d63ac004ad18734c19dc34dbcb36a6772d/hw/drivers/pwm/include/pwm/pwm.h#L112).

mlaz, 2018-04-17:
Ok, I agree with the most that have been said here but we need to take a look at the nature of the problem and how PWM is implemented. Typically there's a HW device which is active or inactive, it uses different channels which might have a duty cycle. I think that what we're failing at here is having something called enable_duty_cycle instead of set_duty_cycle. This is because we are in fact enabling and disabling a device(as long as we're enabling at least a single channel or disabling all of them). So IMO, we should have both device configure/enable/disable functions and channel configure/unconfigure/set_duty. Note that we are setting the duty cycle to 0 whenever disabling a channel, which doesn't make that much sense. I would like to see a flow for a basic set up like: pwm_set_frequency, pwm_configure_channel, pwm_set_duty, pwm_enable.

mlampert, 2018-05-08:
Where are we on the api change? Is the current API stable?

mlaz, 2018-05-08:
Hi Markus, the API and both drivers' refactors are ready for PR, I am changing this on most BSP's right now.

mlampert, 2018-05-08:
Assuming `pwm_enabled` is the branch for the PR.

I had a look and like it, makes a lot of sense. Naturally I have some questions (didn't look at your implementations in order to not taint the API review):

* The cycle and sequence have moved from the channel to the device. This simplifies things quite a bit and makes a lot of sense for the cycle. I am wondering though if it would make sense to keep the sequence on the channel. I don't have a real world use case either way, but I could imagine that I want to keep one channel going (forever) while running sequences on other channels of the same device. This is more an ask for "inspiration" ;)

* I presume the general workflow is:
  1) os_dev_open
  2) pwm_configure_device (is this required? if not used, can it be skipped?)
  3) pwm_configure_channel (for each channel to be used)
  4) pwm_set_duty_cycle (for each channel to be use)
  5) pwm_enable (at which points there will actually be an output)
What's the expected behaviour if the pwm_is already "enabled" and one calls
 -) pwm_set_duty_cycle on an already configured and playing channel
 -) pwm_set_duty_cycle for a channel that hadn't been playing so far (but is properly configured)
 -) I presume calling pwm_configure_channel has no impact except that the configured output pin assumes the value corresponding to "inverted"

* when client call "pwm_disable", should the pwm complete the current cycle or immediately terminate the pwm output? Or is that left to the implementation?

* is pwm_get_resolution_bits still needed now that we have pwm_get_top_value ?

* calling pwm_configure_device on a device that has already been configured previously maintains the existing channel configurations?

mlaz, 2018-05-09:
> The cycle and sequence have moved from the channel to the device. This simplifies things quite a bit and makes a lot of sense for the cycle. I am wondering though if it would make sense to keep the [...]

I believe this is the most natural way to implement this feature, typically pwm devices are clock dividers, which may trigger these interrupts. A device has one or more channels, so these channels are all affected by such interrupts. A real world case is for example RGB LED's with the nRF52 PWM driver on which we may associate a single(3 of its channels) device to each LED.

> pwm_configure_device (is this required? if not used, can it be skipped?)

It is not required as you may see on the pwm_test app's code. We should always have a default configuration for the driver on which these features aren't in use.

> when client call "pwm_disable", should the pwm complete the current cycle or immediately terminate the pwm output? Or is that left to the implementation?

I think it makes sense for it to finish the cycle, although this might be a bit too hardware specific. We can force it to finish the cycle on soft PWM but on an hardware based solution it will be up to the driver's implementation.

> calling pwm_configure_device on a device that has already been configured previously maintains the existing channel configurations?

It should.

> is pwm_get_resolution_bits still needed now that we have pwm_get_top_value ?

I'm not sure but it isn't hurting anyone, is it?

mlampert, 2018-05-09:
Thanks for the clarifications.

Any chance I can persuade you to make `pwm_chan_cfg::pin`  a `uint32_t`?

mlaz, 2018-05-09:
Ok, give me a use case where you'd need it. It is uint8_t because gpio hal
uses this data type, which makes sense to me since I doubt you'll have more
than 255 gpio pins on an MCU, besides the coherence between APIs.

mlampert, 2018-05-09:
>> Any chance I can persuade you to make pwm_chan_cfg::pin a uint32_t?
> Ok, give me a use case where you'd need it. It is uint8_t because gpio hal
> uses this data type, which makes sense to me since I doubt you'll have more
> than 255 gpio pins on an MCU, besides the consistency between APIs we want to keep.

For STM32 processors IO is addressed in ports (A,B,...) and each port has (up to) 16 IO pins. The GPIO addressing scheme is `(port << 8) | pin`.  Today you get processors with ports up to `K`.

Additionally each IO pin can be used for up to 16 special purposes (PWM channel output among others).

A while back we had a discussion on the mailing list about introducing a pin type. And adding more `uint8_t` interfaces increases the inertia to move into that direction.

mlaz, 2018-05-09:
Note that on nordic devboards (namely nrf52840pdk) there are also multiple ports containing multiple pins, we are still dealing with this by using some addressing scheme like the one you mentioned(actualy this is the standard way tha manufacturer handles this on its SDK), also GPIO pins are usually either input or output configurable.

>A while back we had a discussion on the mailing list about introducing a pin type. And adding more uint8_t interfaces increases the inertia to move into that direction.

I remember that, it looks a lot like a problem mostly(if not only) happening with stm chips. I think if you really want to move on to new type for the GPIO pins you probably should first implement it and push it (which will probably make you refactor a bunch of code in the process). It doesn't make much sense for me now to implement a new pin type out of the blue for a use case I am not even fully aware of. 
I don't think changing the pin type for uint32_t on the PWM API while having GPIO HAL API pins as uint8_t is very consistent. If you want to move in that direction that is a whole different task there, it won't even help you having a bunch of drivers with(ugly dirty) casts which later on you'll have to refactor again. 
Anyway I am not opposing to anything here, I just don't see a real advantage on having this type changed only on the PWM API, mostly because you'll probably change the GPIO pin type in the future(since that's your objective) which will bring us to an inevitable refactor no matter what. I don't see how it increases inertia, I just feel that since software is always being rewritten we should fight for it to be consistently well written, not like an unfinished WIP.

mlampert, 2018-05-10:
On Wed, 09 May 2018 16:59:53 -0700
Miguel Azevedo <notifications@github.com> wrote:

> Note that on nordic devboards (namely nrf52840pdk) there are also
> multiple ports containing multiple pins, we are still dealing with
> this by using some addressing scheme like the one you [...]

But what you want here is not an IO pin, definitely not a GPIO pin, you want a PWM pin. I'm not sure the comparison to GPIO is applicable here, this is new ground.

mlampert, 2018-05-10:
On Wed, 9 May 2018 17:07:28 -0700
markus <markus@bibi.ca> wrote:

> On Wed, 09 May 2018 16:59:53 -0700
> Miguel Azevedo <notifications@github.com> wrote:
>
>
> But what you want here is not an IO pin, definitely not a GPIO pin,
> you want a PWM pin. I'm not sure the [...]

To emphasize, this is not an STM32 specific issue, all Cortex-M processors I'm dealing with (STM, TI, NXP) have some sort of "alternate function" that needs to be specified in order to connect a pin to a peripheral.

In other words, while it is currently possible to specify a GPIO pin with 8 bits (because the alternate function by default and implicitly is 0), it is not possible to specify an IO pin for PWM in 8 bits for any of those processors.

In summary, 8 bit might work for NRF processors, but not for a lot of the other supported mcu's.

mlaz, 2018-05-10:
@mlampert, so far we are doing soft PWM and hw PWM using GPIO pins, because these are the only pins we got an abstraction which represents them (besides the fact that soft PWM uses necessairily GPIO and most, if not all, pins on nRF52xx are GPIO). However you are still asking me to have some out of the pattern type here, with _no benefit_ whatsoever. Please implement these new pin abstraction, ping me and I'll be happy to then refactor this code.

mlaz, 2018-05-10:
> But what you want here is not an IO pin, definitely not a GPIO pin, you want a PWM pin. I'm not sure the comparison to GPIO is applicable here, this is new ground. 

This statement's validity is highly hardware dependent, however I still doubt you'll have more than 255 'PWM pins'. By the way, I was not comparing anything, nrf52 PWM and soft PWM use GPIO pins.

mlampert, 2018-05-10:
On Thu, 10 May 2018 08:59:00 -0700
Miguel Azevedo <notifications@github.com> wrote:

> This statement's validity is highly hardware dependent, however I
> still doubt you'll have more than 255 'PWM pins'. By the way, I was
> not comparing anything, nrf52 PWM and soft PWM use GPIO pins.

Exactly my point. And if I check hal_gpio.h in my repository every single api call uses an int.
But since supporting any other hardware than nrf52 and soft PWM has "no benefit whatsoever" - a uint8_t is all you need.

mlaz, 2018-05-10:
Ok, you are right. I will change it.

mlaz, 2018-05-15:
Solved on https://github.com/apache/mynewt-core/pull/1016

closed 2018-05-15 by mlaz