apache/mynewt-core #965: Conflicting types for stm32 precision table

ccollins476ad, 2018-03-27:
Some C files define `AHBPrecTable` as:
```__I uint8_t AHBPrescTable[16]``` 

(`__I` expands to `volatile const`).

However, the header file declares it as:
```
const uint8_t AHBPrescTable[16]
```

This mismatch causes build errors like the following:
```
Error: repos/apache-mynewt-core/hw/bsp/sensorhub/src/system_stm32f4xx.c:126:15: error: conflicting types for 'AHBPrescTable'
   __I uint8_t AHBPrescTable[16] = {0, 0, 0, 0, 0, 0, 0, 0, 1, 2, 3, 4, 6, 7, 8, 9};
               ^
In file included from repos/apache-mynewt-core/hw/mcu/stm/stm32f4xx/src/ext/Drivers/CMSIS/Device/ST/STM32F4xx/Include/stm32f427xx.h:189:0,
                 from repos/apache-mynewt-core/hw/mcu/stm/stm32f4xx/src/ext/Drivers/CMSIS/Device/ST/STM32F4xx/Include/stm32f4xx.h:153,
                 from repos/apache-mynewt-core/hw/bsp/sensorhub/src/system_stm32f4xx.c:66:
repos/apache-mynewt-core/hw/mcu/stm/stm32f4xx/src/ext/Drivers/CMSIS/Device/ST/STM32F4xx/Include/system_stm32f4xx.h:76:23: note: previous declaration of 'AHBPrescTable' was here
 extern const uint8_t  AHBPrescTable[16];    /*!< AHB prescalers table values */
```

I am not sure if the `volatile` qualifier is necessary or not.

utzig, 2018-03-27:
What's your gcc version?

utzig, 2018-03-27:
Ah, I've seen this warning before when I built all ARM BSPs two weeks ago... sorry, forgot to fix... all other BSPs use `const uint8_t AHBPrescTable[16]` so that should be safe to use.

ccollins476ad, 2018-03-27:
By the way, I am seeing another related issue with the `sensorhub` bsp:

```
Error: /Users/ccollins/tmp/myproj2/bin/targets/blinky-sensorhub/app/hw/mcu/stm/stm32f4xx/hw_mcu_stm_stm32f4xx.a(stm32f4xx_hal_rcc.o): In function `HAL_RCC_GetPCLK1Freq':
/Users/ccollins/tmp/myproj2/repos/apache-mynewt-core/hw/mcu/stm/stm32f4xx/src/ext/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rcc.c:946: undefined reference to `APBPrescTable'
/Users/ccollins/tmp/myproj2/bin/targets/blinky-sensorhub/app/hw/mcu/stm/stm32f4xx/hw_mcu_stm_stm32f4xx.a(stm32f4xx_hal_rcc.o): In function `HAL_RCC_GetPCLK2Freq':
/Users/ccollins/tmp/myproj2/repos/apache-mynewt-core/hw/mcu/stm/stm32f4xx/src/ext/Drivers/STM32F4xx_HAL_Driver/Src/stm32f4xx_hal_rcc.c:958: undefined reference to `APBPrescTable'
```

utzig, 2018-03-27:
All stm32f4 based BSPs declare it like this: `const uint8_t APBPrescTable[8] = {0, 0, 0, 0, 1, 2, 3, 4};`. This is part of the standard `system_stm32f4xx.c` which is provided by stm32cube and we copy to the BSPs. Not sure how `sensorhub` BSP was made, but if just adding this line still give you errors, you should be totally safe copying `system_stm32f4xx.c` from the `stm32f4discovery` BSP.

kasjer, 2021-02-23:
sensorhub is remove and this problem is not longer an issue

closed 2021-02-23 by kasjer