apache/mynewt-core #1027: more than 31 bytes in advertsing and scanning

swapkada, 2018-04-15:
hello, I am using btshell and trying to send more than 31 bytes but not able to send. it shows invalid mfg_data.. but if I put less than 31 bytes in mfg_data it accepts. is there any alternative way to send more than 31 bytes?  is there anything which I am missing?
Thank you in advance

SharanIoT, 2018-05-07:
@swapkada Advertising or Scan Response data length is max 31 Octets as per the Bluetooth Core Spec 4.2

sjanc, 2018-05-16:
Hi,

You can use Advertising Extensions which allows for up to 1650 bytes of data. Check BLE_EXT_ADV and BLE_EXT_ADV_MAX_SIZE syscfg values.

You need scanner that supports extended advertising to be able to see those, NimBLE supports it. Also some latest top Android Phones (eg Sony Xperia XZ2) support those.

closed 2018-05-16 by sjanc