apache/mynewt-core #1082: Why don't look for the NimBLE source code?

panuniverse, 2018-05-11:


andrzej-kaczmarek, 2018-05-11:
what is exactly the problem here?

panuniverse, 2018-05-11:
Hi, I select the master branch which I don't look for the NimBLE Source Code. While switching the develop branch, I could find the NimBLE Source Code. It's my fault.
@andrzej-kaczmarek

andrzej-kaczmarek, 2018-05-11:
NimBLE code is moved to separate repository - https://github.com/apache/mynewt-nimble
If you're using `master` branch of `mynewt-core` you need to do `newt upgrade` in your project and it will fetch missing dependencies (i.e. `mynewt-nimble` repository). For this to work you will also need `newt` build from `master` - `newt` v1.3 will not work.

panuniverse, 2018-05-27:
Thank for your reply.

panuniverse, 2018-05-27:
@andrzej-kaczmarek  Thank for your reply.