apache/mynewt-core #1093: timeout/tick / int32_t vs uint32_t

sdalu, 2018-05-15:
Delay/Timeout are expresed in ticks but sometimes with int32_t sometimes with uint32_t, this is confusing and should be unified

~~~c
os_time_delay(int32_t osticks);
os_callout_reset(struct os_callout *c, int32_t ticks);
os_mutex_pend(struct os_mutex *mu, uint32_t timeout);
os_sem_pend(struct os_sem *sem, uint32_t timeout);
~~~

andrzej-kaczmarek, 2018-05-15:
I'd say it should be changed to `os_time_t` everywhere - some functions already use it. this way we know which parameters accept number of OS ticks as a value.

wes3, 2018-05-15:
I would be more in favor of just making it os_time_t; no need to introduce os_ticks_t as that seems redundant to me.

sdalu, 2018-05-24:
What would be the use for negative time? Seems dangerous to me to have this kind of type.

I only see use for time (expressed from OS start) and delta (ticks) which both are positive number.

With signed time people can be tempted to compute `delta = past - now` which will be undefined behaviour in case of overflow

ccollins476ad, 2018-05-24:
Assigning a too-large number to a signed type is not undefined behavior, it is implementation defined.  gcc (and I believe nearly all compilers) define this behavior in the expected way: https://gcc.gnu.org/onlinedocs/gcc/Integers-implementation.html

So, that is exactly what the signed time type is for :).

ccollins476ad, 2018-05-24:
That page describes something slightly different: operations on signed integers that result in overflow.  That is indeed undefined in the C standard.  However, that does not apply to something like the following:

```
uint32_t now;
uint32_t then;
int32_t delta;

// [...]
delta = now - then;
```

In this excerpt, all operations are performed on unsigned integers.  The resulting unsigned integer is then implicitly converted to a signed integer via assignment.  *Conversion* of out-of-range values to a signed integer is implementation defined; *arithmetic* on signed integers resulting in overflow is undefined.

From the c11 draft specification (for all intents and purposes it is identical to the official c11 standard, it is just free instead of costing hundreds of dollars): http://www.iso-9899.info/n1570.html; this is documented in section 6.3.1p3.

sdalu, 2018-05-24:
Sorry my bad, I should have been more clear, and i realise now why i didn't fully understood your previous  answer.

I was thinking that due to the name, it could be tempting to also use `os_stime_t` for `then` and `now` which would lead to problems. That was also why I like the idea to make distinction for type representing time `os_time_t` and type representing ticks/duration `os_ticks_t` `os_sticks_t`

ccollins476ad, 2018-05-24:
I don't know how I feel about adding `os_ticks_t`; I don't necessarily disagree, but I need to give it more thought.

However, I think you make a very good point.  Improperly comparing time values yields some nasty bugs, and I think Mynewt needs to do more to make time comparisons less error prone.  The problem with this class of bugs is that they remain hidden until a device is up and running for a long time.  Specifically, I'm thinking of code which compares time values like this:
```
if (now >= timer_expiry) { /* ... */ }
```

Since these time values roll over eventually, this type of naive comparison is buggy.  Instead, the user needs to use the `OS_TIME_TICK_[...]` macros (or equivalent).

Do you think the addition of `os_[s]ticks_t` would help prevent bugs like this?  C doesn't help out much here; `os_time_t`, `os_ticks_t`, and `uint32_t` would all be interchangeable.

By the way, the `os_time_t`/`os_stime_t` naming scheme is based on the standard `size_t/ssize_t` types.  It is not logical to have a negative size, but I have not seen people use `ssize_t` inappropriately.

wes3, 2018-05-24:
I probably should have been more emphatic in my original comment. Do not add os_ticks_t!!!!! It will not be helpful at this point and will only add to the confusion.

sdalu, 2018-05-28:
@wes3 can you explain why you find it will only add confusion? 

@ccollins476ad having a (short?) webpage on common pitfalls (time manipulation, stack size, interrupt handling, ...) could perhaps help raising awardness

sdalu, 2018-06-18:
Closing following merge of #1186

closed 2018-06-18 by sdalu