apache/mynewt-core #1096: nimble dependency error

SharanIoT, 2018-05-16:
Hi,

I'm getting the nimble dependency error while building blemesh_light sample from the master branch. I didn't face this error with 'tag_1_3_0'. Couldn't find any useful links how to resolve this issue. Any help would be appreciated.

`Building target targets/blemesh_light
Error: Could not resolve package dependency: @apache-mynewt-nimble/nimble/controller; depender: net/nimble/controller`

andrzej-kaczmarek, 2018-05-16:
to work on `master` branch you will also need `newt` build from `master` and do `newt upgrade` in your repository - this will download all dependencies, i.e. NimBLE which is now places in separate repository (`mynewt-nimble`)

SharanIoT, 2018-05-16:
Sorry, didn't get this - 

> "to work on master branch you will also need newt build from master.

I did 'newt install' and 'newt upgrade' too. It says already "installed and upgraded"

andrzej-kaczmarek, 2018-05-16:
because you use `newt` tool v1.3 which does not have proper support for dependencies in repositories.
you need to download `newt` sources from https://github.com/apache/mynewt-newt and build it. not that this tool is written in Go so it's not a regular makefile project.

rymanluk, 2018-05-16:
Hi,

other way is to edit project.yml file and add there 3 things:

1) under project.repositories:
- apache-mynewt-nimble

2)
repository.apache-mynewt-nimble:
   type: github
   vers: 0-dev
   user: apache
   repo: mynewt-nimble

3) change vers. of mynewt-core to 0-dev

After that do newt upgrade. Hope that helps

\Łukasz

SharanIoT, 2018-05-16:
Thanks andrez and rymanluk. Both of your suggestions work! Appreciate your help.
However, there are some application specific build errors which I can fix myself.

Wonder how this type of build errors can be in the git. Or I'm missing something?

`Assembling repos/apache-mynewt-core/hw/bsp/nrf52dk/src/arch/cortex_m4/gcc_startup_nrf52.s
Error: repos/apache-mynewt-core/apps/blemesh_light/src/light_model.c: In function 'light_set_lightness':
repos/apache-mynewt-core/apps/blemesh_light/src/light_model.c:44:7: error: implicit declaration of function 'pwm_enable_duty_cycle' [-Werror=implicit-function-declaration]
  rc = pwm_enable_duty_cycle(pwm0, 0, pwm_val);
       ^`

andrzej-kaczmarek, 2018-05-16:
This is because PWM API was changed recently on `master` and `blemesh_light` was not updated apparently.

@michal-narajowski can you take a look?

michal-narajowski, 2018-05-16:
Update for `blemesh_light` was just merged: https://github.com/apache/mynewt-core/pull/1099

SharanIoT, 2018-05-17:
Hi Michal,

After pulling your changes the PWM API errors in the blemesh_light are gone. But encountered the following error:

`Error: In file included from repos/apache-mynewt-core/hw/drivers/pwm/pwm_nrf52/src/pwm_nrf52.c:31:0:
repos/apache-mynewt-core/hw/mcu/nordic/src/ext/nrfx/drivers/include/nrfx_pwm.h:123:37: error: 'NRFX_PWM_DEFAULT_CONFIG_COUNT_MODE' undeclared here (not in a function)
     .count_mode   = (nrf_pwm_mode_t)NRFX_PWM_DEFAULT_CONFIG_COUNT_MODE,    \
                                     ^
repos/apache-mynewt-core/hw/drivers/pwm/pwm_nrf52/src/pwm_nrf52.c:58:18: note: in expansion of macro 'NRFX_PWM_DEFAULT_CONFIG'
     [0].config = NRFX_PWM_DEFAULT_CONFIG,`

SharanIoT, 2018-05-17:
Found the problem. There is a copy-paste error in the nrfx52_config.h :)

#ifndef NRFX_PWM_DEFAULT_CONFIG_BASE_CLOCK
#define NRFX_PWM_DEFAULT_CONFIG_COUNT_MODE 0
#endif

Should be:

#ifndef NRFX_PWM_DEFAULT_CONFIG_COUNT_MODE
#define NRFX_PWM_DEFAULT_CONFIG_COUNT_MODE 0
#endif

sjanc, 2018-05-18:
good catch!  This is now fixed on master  https://github.com/apache/mynewt-core/pull/1109

sjanc, 2018-05-22:
I think that one is now resolved, right?

closed 2018-05-24 by sjanc