apache/mynewt-core #1115: Error while connecting to multiple Peripherals

kartik9k, 2018-05-21:
Hi all, I've been trying to connect multiple peripherals to a central module using a modified version of blecent example. However after connecting the first peripheral, while connecting the second, I get an error status BLE_HS_ENOMEM (Operation failed due to resource exhaustion).

I've updated peer_init parameters so that appropriate memory is given to the program. Can someone help me in solving this problem?

TIA!

ccollins476ad, 2018-05-21:
Hi Kartik,

Unfortunately, `BLE_HS_ENOMEM` is heavily overloaded, so it sometimes
takes a bit of trial and error to solve these issues.

My first guess is that your `BLE_MAX_CONNECTIONS` setting is too low.
By default this value is only 1, so you'll definitely need to increase
it.

Chris

kartik9k, 2018-05-21:
@ccollins476ad thanks for a prompt response, I have changed the call to the peer_init() function to peer_init(5, 20, 20, 20) which should be enough for 2 peripherals, I guess.

Is there anything else that I'm missing or is it just trial and error?

If possible could you share a working code of a central module which connects to more than 1 peripheral?

ccollins476ad, 2018-05-21:
`BLE_MAX_CONNECTIONS` is used in several places throughout the BLE stack, not just in `peer_init()`.  So, you should not change the call to `peer_init()`.  Instead, you should change the value of `BLE_MAX_CONNECTIONS`.  You can do this by modifying your target's `syscfg.yml` file or with the following newt command:
```
newt target amend <target-name> syscfg=BLE_MAX_CONNECTIONS=4
```

(to increase the max connections to 4).

kartik9k, 2018-05-21:
Hi @ccollins476ad thanks for the insight! It works perfectly now!

Thanks!

closed 2018-05-21 by kartik9k