apache/mynewt-core #1145: Can't program central module via Mac OS

kartik9k, 2018-05-28:
Hi! I was able to connect multiple peripherals with some help 2 days ago. I was a linux machine then, now I've switched to Mac OS and am unable to run the same code for central module which ran perfectly fine on Linux 2 days ago.

I am assuming it's related to the gcc version, but not sure

Can anyone tell why is it so?

andrzej-kaczmarek, 2018-05-28:
can you provide some more details what exactly does not work? are you not able to build, flash or it does not run after flashing?

andrzej-kaczmarek, 2018-05-28:
btw, if it does not run try with `newt debug` and provide us backtrace (`bt` command). if it does not build/flash then provide appropriate log with an error.

kartik9k, 2018-05-28:
Hi! The programs builds and loads as usual. The device is working properly for other programs such as the nrf52_adc and peripheral app.

My observation(guess) is that it never reaches the `on_sync` callback which calls the `blecent_scan`. What more should I tell you please let me know?
It was working just fine when I compiled the same in Linux 2 days ago.

kartik9k, 2018-05-28:
I can share my code if you want to? Rather, it's in a repository in my profile. It can be found [here](https://github.com/kartik9k/Dots2.0/tree/master/blecent)

kartik9k, 2018-05-28:
Any help would be appreciated. Sorry for the urgency @andrzej-kaczmarek

kartik9k, 2018-05-29:
As requested, here is the backtrace:
`(gdb) bt
#0  __assert_func (file=file@entry=0x0, line=line@entry=0, 
    func=func@entry=0x0, e=e@entry=0x0)
    at repos/apache-mynewt-core/kernel/os/src/arch/cortex_m4/os_fault.c:137
#1  0x000160a4 in ble_hs_event_start (ev=<optimized out>)
    at repos/apache-mynewt-core/net/nimble/host/src/ble_hs.c:474
#2  0x000089ba in main ()
    at repos/apache-mynewt-core/apps/blecent/src/main.c:372`

andrzej-kaczmarek, 2018-05-29:
so as you can see it stops on an assert in following function:
```
468 static void
469 ble_hs_event_start(struct os_event *ev)
470 {   
471     int rc;
472     
473     rc = ble_hs_start();
474     assert(rc == 0);
475 }
```
the only reason why `ble_hs_start()` can return an error is when `ble_gatts_start()` returns an error which usually occurs due to out of free memory. so the quickest solution here would be to step through `ble_gatts_start()` with debugger and see where it fails.

also, if the same code works fine on Linux I suspect there is difference in either Mynewt version (on Mac you clearly have 1.3.0) or target configuration (`newt target config show <target>`) so you should also check for differences there.

andrzej-kaczmarek, 2018-05-29:
I think the reason why it fails is following fix missing in 1.3.0: https://github.com/apache/mynewt-core/pull/700 - please apply it and retry.

kartik9k, 2018-05-29:
Hey! Thanks a lot! It works perfectly now :D

I hope this is the only fix missing in 1.3.0

closed 2018-05-29 by andrzej-kaczmarek