apache/mynewt-core #1173: Awkward log API requires padded buffer

ccollins476ad, 2018-06-07:
The `log_append()` function writes the contents of a flat buffer to the specified log.  However, the provided buffer must contain `LOG_ENTRY_HDR_SIZE` bytes of padding at the beginning, since the function populates this space with the log entry header.  So, to log the string "abc", you must pass the following arguments to this function:
* `data`: `<padding>abc`   (total of `LOG_ENTRY_HDR_SIZE`+3 bytes.)
* `len`: 3

Similarly, `log_append_mbuf()` requires that the mbuf contain `LOG_ENTRY_HDR_SIZE` of padding at the start.  The provided mbuf must have these characteristics:
* `om_data`: `<padding>abc`
* `om_len`: `LOG_ENTRY_HDR_SIZE`+3

I understand why the API is like this - some underlying storage media, namely FCB, requires writes to be done in a single chunk.  However, I think this API is too awkward and error-prone to endure.  Perhaps we need to modify the fcb code to allow split writes.

mkiiskila, 2018-06-08:
FCB does allow split writes; but note that flash_align will need to be paid attention to when doing it.

ccollins476ad, 2018-06-12:
Thanks for the correction, Marko.  That makes sense.

I propose the following changes:
1. Modify the log entry header such that the `ue_level` field is split into two fields: "level" and "flags" (one nibble each).
2. Define a `LOG_ENTRY_HDR_F_PADDED` flag.
3. If this new flag is set in an entry, it indicates that the log entry body begins on the next flash-aligned address, rather than immediately after the log header.
4. Change the semantics of the `log_append{,_typed,_mbuf}` functions such that the provided payload does not contain room for a header.  Instead, these function define the entry header on the stack, fill it, and write it separately.

Notes:
* The over-the-air format of log commands and responses would not change.
* I chose to split the `ue_level` field because I don't think 16 levels is plenty.  I also don't think it makes too much sense to allow an application to define custom "per-user" levels, since levels be monotonically increasing.  I.e., all custom levels would be more severe than "critical".

To maintain backwards compatibility, we would need to either:
1. Create newly named append functions with the new semantics, or
2. Introduce a new log version or new syscfg setting to enable the new behavior.

Thoughts?

ccollins476ad, 2018-06-13:
Looking into this a bit more, I discovered a few things:

1. The `log_append_mbuf` and `log_append_mbuf_typed` functions assume a flash write size of 1.  So these won't work on some hardware.
2. A lot of our system relies on the fact that the header and body are adjacent in a log entry.  So my idea of inserting padding as needed after the entry header would cause quite a headache.

Re: 2- Specifically, the `log_read()` function fills a buffer with data from the log, starting with the log entry header.  The caller then skips past the header in the buffer to access the message body.  This is problematic if there might be padding after the header.  This issue also affects log walks; a typical walk callback reads the current entry using `log_read()`.

We could add a "header-size" callback to `struct log_handler`.  Using this, the called could know where the log body begins.

Alternatively, the `log_read()` function could return two pieces of data:
1. The entry header (`struct log_entry_hdr`)
2. Part of the message body (length equal to number of bytes requested)

So, the caller would specify the number of body bytes to read, and would always get back the full header and the specified number of body bytes.

ccollins476ad, 2018-06-13:
I thought of an obvious solution to this problem:

1. The `log_append` functions define a small buffer on the stack, with size equal to `sizeof (struct log_entry_hdr)` rounded up to the nearest min-write-size multiple.

2. The `log_append` functions split their write into two phases:
a. header + first-x bytes of the body
b. remaining bytes in the message body

This doesn't address the mbuf issue, but I don't think there is any easy way around that.  I think the right solution in this case is to require a min-write-size of 1 for the `append_mbuf` API.

andrzej-kaczmarek, 2018-06-13:
as for the mbuf issue, of the top of my head, we may rearrange data on the fly, i.e. write as much as possible from 1st mbuf in chain, then assemble small aligned chunk using remaining data and few bytes from beginning of 2nd mbuf in chain and so on. similar to what is done in no 2 above.

ccollins476ad, 2018-06-13:
Andrzej, that sounds good, though it might be a bit tricky.  Perhaps it can be addressed in a later PR :).

PR #1196 (2018-06-22): sys/log/full: Add "body" functions to API
This fixes #1173.

This PR adds five functions to the log API:
* `log_append_body`
* `log_append_mbuf_body`
* `log_read_hdr`
* `log_read_body`
* `log_read_mbuf_body`
* `log_walk_body`

The "append" functions are similar to `log_append_typed` and `log_append_mbuf_typed`.  The difference is that the provided flat buf or mbuf does not need to contain padding at the beginning for holding a log entry header.

`log_read_hdr` just reads the specified entry header from the log.

The "read body" functions are very similar to the existing "read" function.  The o [... cut]
commit 101f7c47 (2018-06-22): sys/log/full: Add "read_body" functions to API

This fixes #1173.

This commit adds three functions to the log API:
* `log_read_hdr`
* `log_read_body`
* `log_read_mbuf_body`

`log_read_hdr` just reads the specified entry header from the log.

The to "read body" functions are very similar to the existing "read"
function.  The only difference is that the specified offset starts from
the beginning of the message body, not the from the beginning of the
entry.  That is, the user doesn't need to add `LOG_ENTRY_HDR_SIZE` to
the offset that gets passed in.
commit 782276d1 (2018-06-22): sys/log/full: Add "append_body" functions to API

This fixes #1173.

This commit adds two functions to the log API:
* `log_append_body`
* `log_append_mbuf_body`

These functions are similar to `log_append_typed` and
`log_append_mbuf_typed`.  The difference is that the provided flat buf
or mbuf does not need to contain padding at the beginning for holding a
log entry header.
closed 2018-06-22 by ccollins476ad