apache/mynewt-core #1178: repos/apache-mynewt-core/boot/boot_serial/test/src/boot_test.h:22:20: fatal error: assert.h: No such file or directory

wnnwoo, 2018-06-09:
When I setup the newt env as the guide in this link in Mac OS.

https://mynewt.apache.org/latest/os/get_started/project_create/

and there will be **test fail** when I run the command: newt test all, who can help me?

Here is some the fail log, **it seems some header file path has been changed.**

Mac01:myproj wnnwoo$ newt test all > err_2018_06_09.log
Error: Test failure(s):
Passed tests: []
Failed tests: [boot/boot_serial/test boot/bootutil/test crypto/mbedtls/test encoding/base64/test encoding/cborattr/test encoding/json/test fs/fcb/test fs/nffs/test kernel/os/test net/ip/mn_socket/test net/nimble/controller/test net/nimble/host/test net/oic/test sys/config/test-fcb sys/config/test-nffs sys/flash_map/test sys/log/full/test util/cbmem/test]
Mac01:myproj wnnwoo$ ls
LICENSE			README.md		bin			project.state		repos
NOTICE			apps			err_2018_06_09.log	project.yml		targets
Mac01:myproj wnnwoo$ vim err_2018_06_09.log 
Mac01:myproj wnnwoo$ 
Mac01:myproj wnnwoo$ vim err_2018_06_09.log 

Testing package @apache-mynewt-core/boot/boot_serial/test
Compiling repos/apache-mynewt-core/boot/boot_serial/src/boot_uart.c
Compiling repos/apache-mynewt-core/boot/boot_serial/test/src/testcases/boot_serial_empty_img_msg.c
Compiling repos/apache-mynewt-core/boot/boot_serial/src/boot_serial.c
Compiling repos/apache-mynewt-core/boot/boot_serial/test/src/testcases/boot_serial_empty_msg.c
repos/apache-mynewt-core/boot/boot_serial/src/boot_uart.c:19:20: fatal error: assert.h: No such file or directory
compilation terminated.
Testing package @apache-mynewt-core/boot/bootutil/test
Compiling repos/apache-mynewt-core/boot/bootutil/src/image_ec256.c
Compiling repos/apache-mynewt-core/boot/bootutil/src/image_ec.c
Compiling repos/apache-mynewt-core/boot/bootutil/src/image_rsa.c
Compiling repos/apache-mynewt-core/boot/bootutil/src/bootutil_misc.c
repos/apache-mynewt-core/boot/bootutil/src/image_ec256.c:20:20: fatal error: string.h: No such file or directory
compilation terminated.
Testing package @apache-mynewt-core/crypto/mbedtls/test
Compiling repos/apache-mynewt-core/crypto/mbedtls/src/arc4.c
Compiling repos/apache-mynewt-core/crypto/mbedtls/src/asn1parse.c
Compiling repos/apache-mynewt-core/crypto/mbedtls/src/aes.c
Compiling repos/apache-mynewt-core/crypto/mbedtls/src/aesni.c
In file included from /usr/local/Cellar/gcc/5.3.0/lib/gcc/5/gcc/x86_64-apple-darwin15.0.0/5.3.0/include-fixed/syslimits.h:7:0,
                 from /usr/local/Cellar/gcc/5.3.0/lib/gcc/5/gcc/x86_64-apple-darwin15.0.0/5.3.0/include-fixed/limits.h:34,
                 from repos/apache-mynewt-core/crypto/mbedtls/include/mbedtls/check_config.h:36,
                 from repos/apache-mynewt-core/crypto/mbedtls/include/mbedtls/config.h:2717,
                 from repos/apache-mynewt-core/crypto/mbedtls/src/arc4.c:28:
/usr/local/Cellar/gcc/5.3.0/lib/gcc/5/gcc/x86_64-apple-darwin15.0.0/5.3.0/include-fixed/limits.h:168:61: fatal error: limits.h: No such file or directory
compilation terminated.
Testing package @apache-mynewt-core/encoding/base64/test
Compiling repos/apache-mynewt-core/encoding/base64/src/hex.c
Compiling repos/apache-mynewt-core/encoding/base64/test/src/testcases/str2hex.c
Compiling repos/apache-mynewt-core/encoding/base64/test/src/testcases/hex2str.c
Compiling repos/apache-mynewt-core/encoding/base64/src/base64.c
repos/apache-mynewt-core/encoding/base64/src/hex.c:20:22: fatal error: inttypes.h: No such file or directory
compilation terminated.

mkiiskila, 2018-06-26:
This are errors when trying to build C-code natively to run on MacOS. Have you installed Xcode? That's what should come with these headers. You might also need to install the command line tools for it.

wnnwoo, 2018-07-02:
Yes，I have installed Xcode

closed 2018-07-02 by wnnwoo

wnnwoo, 2018-07-02:
The version is Version 9.4.1 (9F2000)

lunarpulse, 2018-07-18:
Same problem here on a linux system.

utzig, 2018-07-18:
@lunarpulse You're most probably missing the base devel package on your disto.