apache/mynewt-core #1187: Disconnecting while writing analog value from Central Module

kartik9k, 2018-06-12:
Hey all! I'm trying to write analog sensor data from my central module to my peripheral module (the setup is so due to some architecture that I'm trying to implement).

I can receive the values from the ADC callback perfectly, but I'm not able to write them to the peripheral module. The peripheral module disconnects right when ```ble_gattc_write_flat``` command is called.

Here is my edited code for ADC call back for reference: 
```int adc_read_event(struct adc_dev *dev, void *arg, uint8_t etype,
        void *buffer, int buffer_len)
{
    int value;
    int rc;

    value = adc_read(buffer, buffer_len);
    if (value >= 0) {
        console_printf("Got %d\n", value);
    } else {
        console_printf("Error while reading: %d\n", value);
        goto err;
    }
    hal_gpio_toggle(led_pin);
     rc = ble_gattc_write_flat(peer_array[0]->conn_handle, chr_array[0]->chr.val_handle,
                               &f, sizeof f, blecent_on_write, NULL);
     if (rc != 0) {
         BLECENT_LOG(ERROR, "Error: Failed to write characteristic; rc=%d\n",
                     rc);
     }        
    return (0);
err:
    return (rc);
} ```

Is it something to do with the ```ble_hs_lock()``` and the ```ble_hs_unlock()``` which are used in ```ble_gatts_chr_updated``` from the original example.

Any help is appreciated. 

Thanks! :d

ccollins476ad, 2018-06-12:
That code looks good to me.  Do you know the reason for the disconnect?  The gap event callback should indicate the reason in `disconnect.reason`.

kartik9k, 2018-06-12:
Thanks for your prompt reply. The code, however, never reaches the Disconnect callback in the gap event callback!

kartik9k, 2018-06-12:
@ccollins476ad sorry for annoying you. But any insights would be really appreciated! I'm stuck here since quite some time! Do you this it's related to the ``ble_hs_lock()``` somehow?

andrzej-kaczmarek, 2018-06-12:
if it does not reach disconnect callback then it probably crashed - you can connect debugger and check what happened (or post backtrace and we'll see).

also I would not recommend to call notify directly from interrupt context (I think ADC callback is called from interrupt context) since it takes some time to process it. instead, create an event which you can post to default event queue and send notification from there.

kartik9k, 2018-06-12:
Here's the BT log. Couldn't make sense out of it, myself.
```
#0  0x0000b1a8 in os_time_delay (osticks=osticks@entry=1)
    at repos/apache-mynewt-core/kernel/os/src/os_time.c:141
#1  0x0002162c in console_queue_char (uart_dev=0x20002ae8 <os_bsp_uart0>, 
    ch=<optimized out>)
    at repos/apache-mynewt-core/sys/console/full/src/uart_console.c:83
#2  0x000216ee in console_out (c=56)
    at repos/apache-mynewt-core/sys/console/full/src/uart_console.c:147
#3  0x00021248 in console_write (str=0x2000fdbf "8\004", cnt=cnt@entry=1)
    at repos/apache-mynewt-core/sys/console/full/src/console.c:109
#4  0x0000c02c in stdout_write (fp=<optimized out>, bp=<optimized out>, n=1)
    at repos/apache-mynewt-core/libc/baselibc/src/mynewt.c:32
#5  0x0000bb32 in fwrite (stream=0x2000017c <_stdin>, nmemb=1, size=1, 
    buf=0x2000fdbf)
    at repos/apache-mynewt-core/libc/baselibc/include/stdio.h:59
#6  fputc (f=0x2000017c <_stdin>, c=<optimized out>)
    at repos/apache-mynewt-core/libc/baselibc/include/stdio.h:75
#7  putf (putp=putp@entry=0x2000017c <_stdin>, c=<optimized out>)
    at repos/apache-mynewt-core/libc/baselibc/src/tinyprintf.c:137
#8  0x0000bc3e in putchw (putp=putp@entry=0x2000017c <_stdin>, 
    p=p@entry=0x2000fe00)
    at repos/apache-mynewt-core/libc/baselibc/src/tinyprintf.c:187
#9  0x0000be02 in tfp_format (putp=0x2000017c <_stdin>, 
    fmt=fmt@entry=0x2bb54 "[ts=%lussb, mod=%u level=%u] ", va=...)
---Type <return> to continue, or q <return> to quit---
    at repos/apache-mynewt-core/libc/baselibc/src/tinyprintf.c:302
#10 0x0000bf0a in vfprintf (f=<optimized out>, 
    fmt=fmt@entry=0x2bb54 "[ts=%lussb, mod=%u level=%u] ", va=...)
    at repos/apache-mynewt-core/libc/baselibc/src/tinyprintf.c:355
#11 0x000251d8 in vprintf (
    format=format@entry=0x2bb54 "[ts=%lussb, mod=%u level=%u] ", ap=..., 
    ap@entry=...) at repos/apache-mynewt-core/libc/baselibc/src/vprintf.c:10
#12 0x00021552 in console_printf (fmt=0x2bb54 "[ts=%lussb, mod=%u level=%u] ")
    at repos/apache-mynewt-core/sys/console/full/src/console_fmt.c:55
#13 0x00021a56 in log_console_append (log=<optimized out>, buf=0x2000fe94, 
    len=47) at repos/apache-mynewt-core/sys/log/full/src/log_console.c:40
#14 0x000218b8 in log_append (log=log@entry=0x200044d8 <ble_hs_log>, 
    module=module@entry=4, level=level@entry=1, data=0x2000fe94, 
    data@entry=0x2000fe8c, len=33)
    at repos/apache-mynewt-core/sys/log/full/src/log.c:250
#15 0x00021924 in log_printf (log=0x200044d8 <ble_hs_log>, 
    module=module@entry=4, level=level@entry=1, 
    msg=0x2b408 "GATT procedure initiated: %s")
    at repos/apache-mynewt-core/sys/log/full/src/log.c:275
#16 0x00017226 in ble_gattc_log_proc_init (name=<optimized out>)
    at repos/apache-mynewt-core/net/nimble/host/src/ble_gattc.c:490
#17 0x000172ac in ble_gattc_log_write (att_handle=att_handle@entry=18, 
    len=<optimized out>, expecting_rsp=expecting_rsp@entry=1)
---Type <return> to continue, or q <return> to quit---
    at repos/apache-mynewt-core/net/nimble/host/src/ble_gattc.c:598
#18 0x0001790e in ble_gattc_write (conn_handle=conn_handle@entry=2, 
    attr_handle=attr_handle@entry=18, 
    txom=0x20001280 <os_msys_init_1_data+584>, 
    cb=cb@entry=0x8651 <blecent_on_write>, cb_arg=cb_arg@entry=0x0)
    at repos/apache-mynewt-core/net/nimble/host/src/ble_gattc.c:3710
#19 0x0001796e in ble_gattc_write_flat (conn_handle=<optimized out>, 
    attr_handle=<optimized out>, data=data@entry=0x2000ffaf, 
    data_len=data_len@entry=1, cb=cb@entry=0x8651 <blecent_on_write>, 
    cb_arg=cb_arg@entry=0x0)
    at repos/apache-mynewt-core/net/nimble/host/src/ble_gattc.c:3760
#20 0x0000860e in adc_read_event (dev=<optimized out>, arg=<optimized out>, 
    etype=<optimized out>, buffer=<optimized out>, buffer_len=4)
    at apps/multirole_analog/src/main.c:130
#21 0x00022842 in nrf52_saadc_event_handler (event=<optimized out>)
    at repos/mynewt_nordic/hw/drivers/adc/adc_nrf52/src/adc_nrf52.c:73
#22 0x000249ce in SAADC_IRQHandler ()
    at repos/mynewt_nordic/hw/mcu/nordic_sdk/src/ext/nRF5_SDK_11.0.0_89a8197/components/drivers_nrf/saadc/nrf_drv_saadc.c:93
#23 <signal handler called>
#24 0x00000000 in ?? ()
#25 0x0000812c in .bss_zero_loop ()
    at repos/apache-mynewt-core/hw/bsp/rb-nano2/src/arch/cortex_m4/gcc_startup_n---Type <return> to continue, or q <return> to quit---
rf52.s:180
```

kartik9k, 2018-06-12:
Also, I did not understand the second part of your comment.

andrzej-kaczmarek, 2018-06-12:
there is log message printed to console from NimBLE code which causes a deadlock since `ble_gattc_write_flat` is called from an interrupt context. basically UART buffer is filled and write routine waits for it to be flushed but this does not happen since UART interrupt cannot preempt ADC interrupt.

as I wrote in previous comment, you should not call `ble_gattc_write_flat` from interrupt context here but send an event to event queue and handle this in task context. see https://mynewt.apache.org/latest/os/tutorials/event_queue/ - "interrupt events" section there describes such case.

ccollins476ad, 2018-06-12:
From the stack trace, it appears Andrzej's suspicions were correct: the problem is that you are calling into the BLE host from within an interrupt.  That is a bad idea in general, but in this case, the crash occurs because the bluetooth host attempts to log to the console during the write procedure.  Logging to the console uses quite a bit of stack space, so it is likely that the interrupt stack is overflowing here.

The solution is as Andrzej indicated: post an event to the default eventq.

This isn't entirely trivial in this case.  You'll need to associate the characteristic data with the event so that your event callback knows what payload to pass to the `ble_gattc_write_flat()` function.  If you don't need to perform multiple writes in rapid succession, then you can just use a single global event and a single global buffer.  However, you will still need to make sure your buffer doesn't become corrupted by a subsequent ADC interrupt while you are in mid-write.

Ignoring the race condition for now, you might do something like this (untested):

```
// On ADC interrupt:
// 1. Write ADC data to global buffer `chr_array`
// [...]
// 2. Post an event to the default event queue:
    os_eventq_put(os_eventq_dflt_get(), &my_notify_event); 
```

Elsewhere, you would define and initialize the event:
```
static void
my_notify_ev_fn(struct os_event *ev)
{
    int rc;
    rc = ble_gattc_write_flat(peer_array[0]->conn_handle, chr_array[0]->chr.val_handle,
                               &value, sizeof value, blecent_on_write, NULL);
    if (rc != 0) {
         BLECENT_LOG(ERROR, "Error: Failed to write characteristic; rc=%d\n",
                     rc);
    }        
}

static struct os_event my_notify_event = {
    .ev_cb = my_notify_ev_fn,
};

kartik9k, 2018-06-12:
Are you suggesting that we should add an event to the default event list in the ADC interrupt?

The pseudocode should be something like @ccollins476ad wrote:
```
In ADC interrupt, add the value to global variable and add the task of sending in the interrupt itself
```
Is my interpretation correct?

ccollins476ad, 2018-06-12:
Yes, you got it.  Call `os_eventq_put()` from within the ADC interrupt handler.

kartik9k, 2018-06-12:
Thank you guys! It's working now!

Keep up the amazing work.

closed 2018-06-12 by kartik9k