apache/mynewt-core #1215: Switching from XTAL_32768 to XTAL_RC

kartik9k, 2018-06-22:
Hey! We were planning to make our own PCBs for a project using SMD versions of BLE Nano 2 (MB-N2). We found that the oscillator LFCLKSRC is optional in the design as per the datasheet provided by Nordic. 

However we tried building and running our applications using ```XTAL_32768: 0``` and ```XTAL_RC: 1```, disabling the LFCLKSRC and using the other oscillator, however, the program does not seem to work fine when built under these conditions.

Is it possible to program the same using XTAL_RC or not?

andrzej-kaczmarek, 2018-06-22:
What exactly does not work?

If you see problems with BLE this is because LFRC has poor accuracy and needs to be calibrated. It can achieve 250ppm (when calibrated every 8 secs) which is still quite poor by BLE standards, but within limits. Such calibration is not implemented at the moment in Mynewt.
For designs without LFXO you can use LFSYNTH which has great accuracy, but requires HFXO to be turned on all the time so this will draw extra power.
For designs with BLE it is recommended to use LFXO to save power (HFXO is only running when needed) and have good accuracy (depending on crystal used).

andrzej-kaczmarek, 2018-06-22:
You need to set:
`XTAL_32768_SYNTH: 1` to enable LFSYNTH
`BLE_XTAL_SETTLE_TIME: 0` to make HFXO be running all the time

Here's more information about configuring LP clock for NimBLE controller: https://mynewt.apache.org/network/ble/ble_setup/ble_lp_clock/

kartik9k, 2018-06-22:
Thank you so much! It has worked now, hoping that it continues.

closed 2018-06-22 by kartik9k