apache/mynewt-core #1287: blinky demo for STM32F4-Discovery not works on STM32F429I-Discovery

mintisan, 2018-07-22:
I followed the steps on [Blinky, your “Hello World!”, on STM32F4-Discovery](https://mynewt.apache.org/latest/tutorials/blinky/blinky_stm32f4disc.html).

And every step's result looks fine.

But doesn't work at all, even I try to reset the board.

I also tried to flash the binary directly with command below:
```
st-flash write /Users/linjinhui/Github/mynewt/blinking/bin/targets/stm32f4disc_blinky/app/apps/blinky/blinky.elf.bin 0x8000000
```

But I can blink the leds with mbed demo or STM32Cube via `st-link`. 

So the hardware is OK.

Versions:
- Apache Newt version: 1.4.1
- masOS 10.11.6
- Open On-Chip Debugger 0.10.0
-  st-flash --version v1.4.0
- arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 7-2017-q4-major) 7.2.1 20170904 (release) [ARM/embedded-7-branch revision 255204]

Do I skip any important steps?

mintisan, 2018-07-22:
I tried to use `stm32f429discovery ` to re-build, but I can't load successfully.

```
➜  blinking newt load stm32f429disc_blinky
Loading app image into slot 1
Error: Downloading /Users/linjinhui/Github/mynewt/blinking/bin/targets/stm32f429disc_blinky/app/apps/blinky/blinky.img to 0x8020000
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
adapter speed: 2000 kHz
adapter_nsrst_delay: 100
none separate
srst_only separate srst_nogate srst_open_drain connect_deassert_srst
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : Unable to match requested speed 2000 kHz, using 1800 kHz
Info : clock speed 1800 kHz
Error: open failed
in procedure 'init' called at file ".openocd_cmds", line 2
in procedure 'ocd_bouncer'

load - Load application image on to the board for <target-name>

Usage:
  newt load <target-name> [flags]

Flags:
      --extrajtagcmd string   Extra commands to send to JTAG software

Global Flags:
  -h, --help              Help for newt commands
  -j, --jobs int          Number of concurrent build jobs (default 4)
  -l, --loglevel string   Log level (default "WARN")
  -o, --outfile string    Filename to tee output to
  -q, --quiet             Be quiet; only display error output
  -s, --silent            Be silent; don't output anything
  -v, --verbose           Enable verbose output when executing commands
➜  blinking
```

utzig, 2018-07-23:
I've never had any issues loading on this particular board but some lower-speed boards sometimes have a similar fail when loading the app (never the bootloader!). It never bothered me enough to try to find the exact error which I would suspect might be on the `openocd_load` function in `hw/scripts/openocd.sh` related to something the bootloader might have just done (assert + reset maybe). Anyway it usually works in those cases if doing a followup `newt load ...` just after the first one or in worst cases pressing the RESET button in the board just before executing `newt load ...`.

mintisan, 2018-07-23:
@utzig 

Thank you for your reply!

I found that `stm32f429discovery` may have two different versions: `stm32f429disc` and `stm32f429disc1`.

I found that my board's is `stm32f429i_disc1 ` but not `stm32f429i_disc`.

So I changed openocd's cfg from `board/stm32f429discovery.cfg` to `board/stm32f429disc1.cfg`. And can be downloaded now.

But the LED isn't blinking either.

Maybe initialization for `stm32f429disc1` board is not supported yet?

utzig, 2018-07-23:
The one I have actually is the `stm32f429disc1`. I guess the only difference between them is the st-link version, no? You can try to find out if a breakpoint/fault happened by `newt debug stm32f429disc_blinky` or `newt debug stm32f429disc_boot` (depending on which one failed). A `backtrace` there might be helpful to find out what's happening. Also you could try joining the slack channel (https://join.slack.com/mynewt/shared_invite/MTkwMTg1ODM1NTg5LTE0OTYxNzQ4NzQtZTU1YmNhYjhkMg), that would allow for a better communication if debugging is required.

sjanc, 2019-08-02:
is this issue stil valid?