apache/mynewt-core #1288: can't load binary to  hifive1 board

mintisan, 2018-07-22:
I can't load binary to board which compiled by `riscv32`. 

I didn't know that if It must be compiled with `riscv64` or not?

Step1: add target type for [hifive1](https://www.sifive.com/products/hifive1/)

```
➜  blinking newt target show
targets/hifive1_blinky
    app=apps/blinky
    bsp=@apache-mynewt-core/hw/bsp/hifive1
    build_profile=debug
targets/hifive1_boot
    app=@apache-mynewt-core/apps/boot
    bsp=@apache-mynewt-core/hw/bsp/hifive1
    build_profile=optimized
```

Step2: build successfully after mofidying the hifive1's `compile.yml` to `riscv32` from `riscv64` :

```
compiler.path.cc: "riscv32-unknown-elf-gcc"
compiler.path.as: "riscv32-unknown-elf-gcc"
compiler.path.archive: "riscv32-unknown-elf-ar"
compiler.path.objdump: "riscv32-unknown-elf-objdump"
compiler.path.objsize: "riscv32-unknown-elf-size"
compiler.path.objcopy: "riscv32-unknown-elf-objcopy"

compiler.flags.base: -std=gnu11 -Wall -Werror -Wno-format-truncation -ffunction-sections -fdata-sections -fno-builtin-printf
compiler.flags.default: [compiler.flags.base, -O1 -ggdb]
compiler.flags.optimized: [compiler.flags.base, -Os -ggdb]
compiler.flags.debug: [compiler.flags.base, -Og -ggdb]

compiler.as.flags: [-x assembler-with-cpp]

compiler.ld.flags: -static -specs=nosys.specs -Wl,--gc-sections -nostartfiles
compiler.ld.resolve_circular_deps: true
compiler.ld.mapfile: true
```

Step3: The load error shows as below:

```
➜  blinking newt load hifive1_boot
Loading bootloader
Error: Downloading /Users/linjinhui/Github/mynewt/blinking/bin/targets/hifive1_boot/app/apps/boot/boot.elf.bin to 0x20000000
Open On-Chip Debugger 0.10.0
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 10000 kHz
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
embedded:startup.tcl:21: Error: Unknown target type riscv, try one of arm7tdmi, arm9tdmi, arm920t, arm720t, arm966e, arm946e, arm926ejs, fa526, feroceon, dragonite, xscale, cortex_m, cortex_a, cortex_r4, arm11, ls1_sap, mips_m4k, avr, dsp563xx, dsp5680xx, testee, avr32_ap7k, hla_target, nds32_v2, nds32_v3, nds32_v3m, or1k, quark_x10xx,  or quark_d20xx
in procedure 'script'
at file "embedded:startup.tcl", line 60
in procedure 'target' called at file "/Users/linjinhui/Github/mynewt/blinking/repos/apache-mynewt-core/hw/bsp/hifive1/riscv_openocd.cfg", line 38
in procedure 'ocd_bouncer'
at file "embedded:startup.tcl", line 21

load - Load application image on to the board for <target-name>

Usage:
  newt load <target-name> [flags]

Flags:
      --extrajtagcmd string   Extra commands to send to JTAG software

Global Flags:
  -h, --help              Help for newt commands
  -j, --jobs int          Number of concurrent build jobs (default 4)
  -l, --loglevel string   Log level (default "WARN")
  -o, --outfile string    Filename to tee output to
  -q, --quiet             Be quiet; only display error output
  -s, --silent            Be silent; don't output anything
  -v, --verbose           Enable verbose output when executing commands
```

Versions:

- Apache Newt version: 1.4.1
- masOS 10.11.6
- Open On-Chip Debugger 0.10.0
- riscv32-unknown-elf-gcc (GCC) 6.1.0

mintisan, 2018-07-22:
Sorry, OpenOCD's issue...

mintisan, 2018-07-22:
But Why Use `riscv64`, but not `riscv32` ?

closed 2018-07-22 by mintisan

nathanmasifive, 2019-04-26:
Hi,
Do I have to do all the READNE.OSX stuff if I'm using mac?
Thanks