apache/mynewt-core #1389: SPI never initialized and no errors (stm32)

5frank, 2018-09-07:
Trying to use SPI_2 on stm32_e407 devboard but the issue might apply to other bsp:s/mcu:s as well.

In app/myapp/syscfg.yml SPI 2 is enabled:
```
syscfg.vals:
    SPI_2_MASTER: 1
```
The problem is that SPI_2_MASTER is used in mcu layer but not in the bsp layer which result in that SPI 2 is never initialize. (no `hal_spi_init(2, ...)` call).

I guess this issue could either be seen as a problem in the bsp layer (i.e. in hw/bsp/olimex_stm32-e407_devboard/src/hal_bsp.c) as it do not to throw a compile time error. i.e. it could have something like:
```
#ifdef SPI_2_MASTER
#error "SPI_2 support not (yet) implemented"
#endif
```
... or a problem in the mcu layer (in hw/mcu/stm/stm32_common/src/hal_spi.c) as there is no check that hal_spi_init() have been called and give no other error.

Perhaps it is better to fix in in the mcu layer, but not sure how to do that at compile time?

kasjer, 2021-02-23:
SPI (and other peripherals) initialization was moved to MCU stm32_periph.c that is common across all ST devices.
This PR #2186 took care of this problem

sjanc, 2023-09-26:
should be fixed in master

closed 2023-09-26 by sjanc