apache/mynewt-core #1515: bbc microbit tutorial or example?

zerog2k, 2018-11-10:
I'm trying to get blinky or any console_printf to work on bbc microbit and not getting any output or led blinking.

I can't seem to find any examples of working bbc_microbit target projects. Would someone be willing to share?

zerog2k, 2018-11-10:
ok, while i dont think the bbc microbit had write protection, im wondering if it requires an erase to load bootloader/app (and furthermore curious if this is required every time?)
(followed this: http://mynewt.apache.org/latest/tutorials/blinky/rbnano2.html#clear-the-write-protection-on-the-flash-memory)

(would thought that the download sequence for openocd in the bsp should take care of this if loading a bootloader image? - assuming that mynewt bootloader would handle app erase functionality?)

mkiiskila, 2018-11-12:
If I recall correctly, BBC microbit did not have a simple LED to drive with a single GPIO, so maybe it's still just working? I don't have microbit at hand, but according to schematics the serial console is going to JTAG chip, so that should show up as a serial device. And you should be able to verify that it comes up ok with your favorite terminal program.
Blinky does not enable console, but slinky does.

Openocd erases the underlying flash when loading. If the flashing does not 'stick', it does not complain. So if you're uncertain whether flash is write protected or not, you should make sure that what you wrote is what's now in the flash.

I can't recall what was going on with RedBear Nano2 to require eraseall cycle. Maybe they had locked out JTAG access?