apache/mynewt-core #1574: hal_gpio_irq_release() passing the wrong value to hal_gpio_irq_disable()

rleberknight, 2018-12-18:
In apache-mynewt-core/hw/mcu/nordic/nrf52xxx-compat/src/hal_gpio.c
apache-mynewt-core/hw/mcu/nordic/nrf52xxx/src/hal_gpio.c
and
apache-mynewt-core/hw/mcu/nordic/nrf51xxx/src/hal_gpio.c

hal_gpio_irq_release()  takes a pin as an argument, and passes that to hal_gpio_find_pin(pin) to get an index into hal_gpio_irqs[].
It then passes that index to hal_gpio_irq_disable(i).
hal_gpio_irq_disable() then passes the index to  hal_gpio_find_pin()

Using a pin to get an index with hal_gpio_find_pin() and then passing that index to hal_gpio_find_pin()
is obviously wrong.

I believe that hal_gpio_irq_disable() wants the pin, not the index.
So the line
 hal_gpio_irq_disable(i) 
should be changed to 
hal_gpio_irq_disable(pin)

rleberknight, 2018-12-19:
@kasjer I tried pushing a branch to git@github.com:apache/mynewt-core.git  and it failed:

git push origin hal_gpio_irq_release-use-pin-num
ERROR: Permission to apache/mynewt-core.git denied to rleberknight.
fatal: Could not read from remote repository.

utzig, 2018-12-20:
@rleberknight You have to push the branch to your fork, git@github.com:rleberknight/mynewt-core.git, not `apache`.

closed 2019-01-17 by kasjer