apache/mynewt-core #1667: NRF52 HAL: Possible SPI lockup issue if switching between blocking and non-blocking API

wes3, 2019-02-28:
There is code in hal_spi_txrx() that calls hal_spi_stop_transfer(). This code presumes that if you call TASKS_STOP when EVENTS_STOPPED is 0 you will always see EVENTS_STOPPED getting set to 1. I do not believe this is valid in all cases. I think that EVENTS_STOPPED will only get set if TASKS_START was called prior and EVENTS_STARTED is generated.

wes3, 2019-03-01:
Indeed, calling hal_spi_stop_transfer() in hal_spi_txrx() could lock up the SPI. A fix for that particular issue will be committed shortly. However, there are some other issues which need to be addressed. Both of these issues may cause incorrect SPI bytes to be transmitted if the non-blocking driver is used.

1) The current non-blocking API uses the END_START shortcut if the buffer being transferred is > 255 bytes. This will not work if interrupts are disabled or delayed such that the END event occurs before the interrupt handler has time to update the TXD.PTR and and TXD.MAXCNT registers.  If this happens the peripheral will restart the transfer and incorrect bytes will be transmitted. I do not think it possible to guarantee this and thus I think the driver should be rewritten so that it does not use the END_START shortcut. This is not difficult and really the only drawback is that we will have to take the interrupt latency between spi transfers when doing transfers greater than 255 bytes. Note that this applies to the spi master API only (spi slave currently has a 255 byte transfer limit).

2) Anomaly 109 workaround is not present in our hal. This anomaly has to do with DMA possibly sending out an incorrect first byte. I believe that this anomaly should be handled in the non-blocking SPI interface. There will be a syscfg variable that can be turned on/off in order to compile in/out the workaround for this anomaly. Not sure if the default for this variable should be on or off. I am leaning towards off.

andrzej-kaczmarek, 2019-03-01:
@wes3 
1. In current implementation `EVENT_END` does not update pointers, this is done in `EVENT_STARTED` which is called just after transmission was started so they are ready for next transmission right away (registers are double-buffered). The only case I can think of where it could fail is if we have interrupts disabled/delayed for the duration of 2 consecutive transmissions so the `EVENT_STARTED` is not handled before starting 3rd transfer, which is at least ~500us when running at 8 MHz which is quite long. Current implementation was added in https://github.com/apache/mynewt-core/pull/1252/commits/1b53477bf4ffa4ad100a114b4044ccb0453377c7 - previously it worked as you described, i.e. was using `EVENT_END` to update pointers and restart transfer and it caused relatively long delays between consecutive chunks due to latency introduced by interrupt handling. So if we really want old, slower but safe implementation perhaps it's just enough to revert that patch (although I prefer current implementation).
2. We do not need that workaround since we always start SPIM manually thus HFCLK is running and it won't be disabled. The workaround is required only in case SPIM is started from PPI.

wes3, 2019-03-01:
@andrzej-kaczmarek 

1. Yes, I did indeed realize you would have to delay interrupts for that long and that it is unlikely but I think the end result could be pretty catastrophic. I am pretty paranoid about things like this and I think the delays between the 255 byte transfers should be relatively small (unless other interrupts are going). I guess we could raise the priority of this interrupt to help mitigate that. Given 255 bytes at 8 MHz is a 255usec transfer, that interrupt should be pretty quick so I cannot imagine it adding more than 20% to the whole transfer and that is only when large transfers are done (which I do not think are super common). I think that better than possibly writing erroneous things on the flash.

2. That part about the issue only occurring when starting from PPI confused me a bit I have to say. When I read section 3.2 of that document I did see it said "do not use PPI" but then there was some other wording about ensuring the CPU was on 1usec before using the serial engine and 16 usecs after or until transfer is complete. But if simply not using PPI to start this ensures that well all good then.

sterlinghughes, 2019-04-12:
can we close this issue?

wes3, 2019-04-26:
Yes, we can close this.

closed 2019-04-26 by wes3