apache/mynewt-core #1685: Compile error on Mac Mojave

charuwalee, 2019-03-13:
Hi,

I've tried to install mynewt on Mac Mojave version 10.14.3. Every went well until I tried to create a new project. I followed the instructions from https://mynewt.apache.org/latest/get_started/project_create.html and got tens of error messages when I tried to test the project package by using command `newt test @apache-mynewt-core/sys/config`. The snapshot of an example of error messages are as shown in the pictures below.

[image]

[image]

Could anyone give a hint of what went wrong and how to fix this?

Thank you in advance!!
Charuwalee

sjanc, 2019-04-10:
Hi,

You need to use Xcode 9,  I think this is related to deprecation of 32bit binaries (and newt test is using native target which is 32bit)

sterlinghughes, 2019-04-19:
I think we need to keep this open and fix it, a Mojave install breaks newt sim, which most of the docs suggest using.  Also, it's complicated to move back to Xcode 9 and we should at least have instructions to do so.

Also, I'm wondering if we shouldn't take this oppty to use QEMU, or run 32-bit in QEMU.  The support for STM32* etc is there, but pretty weak (and not distributed in homebrew binaries), but we could look at emulating a X86 system by default and running sim in there?

sjanc, 2024-10-29:
fixed in master

closed 2024-10-29 by sjanc