apache/mynewt-core #1755: update queue.h

sdalu, 2019-04-10:
The provided version of queue.h seems to be outdated in functionalities compared to the version available in FreeBSD for example: https://github.com/freebsd/freebsd/blob/master/sys/sys/queue.h

Note, that the FreeBSD version doesn't include CIRCLEQ anymore, as TAILQ is the prefered way to go: https://stackoverflow.com/a/32013537

utzig, 2019-04-10:
Apart from having the latest version, are there any obvious advantages in updating (bug fixes or new data-structures) ?

sdalu, 2019-04-11:
Being able to retrieve the previous element of a double linked list (LIST_PREV) is the obvious missing macro from the current queue.h

Other benefits:
* *_SAFE variants, allowing removal of element during FOREACH
* new manipulation macros: *_SWAP, *_REMOVE_AFTER, *_CONCAT
* debugging support

The idea is not to reinvent the wheel for such common operations

sjanc, 2024-10-29:
better late than never... :)
https://github.com/apache/mynewt-core/pull/3325

sjanc, 2024-10-30:
fixed in master

closed 2024-10-30 by sjanc