apache/mynewt-core #1761: Unable to compile rbnano2_boot

OTLabs, 2019-04-11:
Compiling rbnano2_boot with newt 1.6.0 on MSYS2 I get following error message:

```
user@workstation ~/dev/myproj
$ newt build rbnano2_boot
Building target targets/rbnano2_boot
Compiling repos/apache-mynewt-core/sys/flash_map/src/flash_map.c
Compiling repos/apache-mynewt-core/util/rwlock/src/rwlock.c
Compiling repos/apache-mynewt-core/sys/mfg/src/mfg.c
Error: repos/apache-mynewt-core/util/rwlock/src/rwlock.c: In function 'rwlock_init':
repos/apache-mynewt-core/util/rwlock/src/rwlock.c:195:21: error: missing braces around initializer [-Werror=missing-braces]
     *lock = (struct rwlock) { 0 };
                     ^
repos/apache-mynewt-core/util/rwlock/src/rwlock.c:195:21: error: (near initialization for '(anonymous).mtx') [-Werror=missing-braces]
cc1.exe: all warnings being treated as errors
```

utzig, 2019-04-12:
what's the output of `arm-none-eabi-gcc --version` ?

OTLabs, 2019-04-12:
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 224288]
```

utzig, 2019-04-12:
This was a well know bug which was fixed later on, but 4.9.3 is VERY OLD. If you cannot update to a newer version, a suggested fix would be to update the line to:
```
*lock = (struct rwlock) {{ 0 }};
```
Let me know if it fixes your issue.

utzig, 2019-04-12:
Also, the latest 4.9 release (https://launchpad.net/gcc-arm-embedded/4.9/4.9-2015-q3-update/+download/gcc-arm-none-eabi-4_9-2015q3-20150921-win32.exe) should have the backported fix applied.

OTLabs, 2019-04-12:
I have installed latest 4.9 release (4.9-2015-q3). It has same issue.

```
$ which arm-none-eabi-gcc
/c/Program Files (x86)/GNU Tools ARM Embedded/4.9 2015q3/bin/arm-none-eabi-gcc

$ arm-none-eabi-gcc --version
arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors) 4.9.3 20150529 (release) [ARM/embedded-4_9-branch revision 227977]
```

Is it possible to try non-4.9 release or should we use 4.9 only?

OTLabs, 2019-04-12:
I would like to share with you that I got compilation going with
https://launchpad.net/gcc-arm-embedded/5.0/5-2016-q3-update/+download/gcc-arm-none-eabi-5_4-2016q3-20160926-win32.exe

closed 2019-04-19 by sjanc