apache/mynewt-core #1851: Mynewt install error

asmitajha, 2019-05-31:
Hello,

I recently started with Mynewt OS.
I went through the step by documentation for its installation.
I have successfully installed newt on my Windows10 PC.
I am following the steps as given on your website to create my first project.
Unfortunately, I am getting following erroe o
:

Admin@ZWPL-COM2 MINGW64 ~/dev/myproj
$ newt install
Error: project.yml file specifies nonexistent repo versions:
    apache-mynewt-core,==1.6.0

I tried to change vers in project.yml file but still getting the same error.

and When I am trying to change vers to 0-dev

I am getting error as

Error: cannot normalize version "0-dev" for repo "apache-mynewt-core"; no mapping to numeric version

Kindly help me out to resolve this issue. Hope to hear from you soon.

asmitajha, 2019-06-20:
Things are working fine on Ubuntu

closed 2019-06-20 by asmitajha

haipingjuul, 2019-08-23:
Hi Asmitajha, I am installing Mynewt and have the exact issue you had for Windows 10, for both Newt 1.6.0 and 1.7.0. Can you let me know how you solved the issue?

Thanks 
David

asmitajha, 2019-08-23:
Hello David,
I was unable to fix it on Windows. So, I proceeded with Linux , on Ubuntu it worked perfectly

utzig, 2019-08-23:
On Windows you might install docker desktop, if you haven't already, and get the newt docker image with: `docker pull mynewt/newt:latest`. This should work on Windows and brings `newt` + `newtmgr`.

haipingjuul, 2019-08-26:
Thanks Asmita for letting me know. Someone helped me and it is working, You
need to specify escape=false for Newt install or upgrade commands with
Windows.

David