apache/mynewt-core #1890: OS: Malloc mutex uninitialized

clmklk, 2019-06-21:
Hello,

I am currently having crashes that are due to the malloc mutex which doesn't seem to be initialized.

A solution could be to create a `os_malloc_init` function, initializing the mutex and call this function in `os_init`.

ccollins476ad, 2019-06-21:
I don't understand the issue.  Could you please clarify with an example?

clmklk, 2019-06-21:
The malloc mutex not being initialized, I have sometimes members of the structure `os_mutex` with a value different from 0.

And in the function `os_mutex_release`, the line `rdy = SLIST_FIRST(&mu->mu_head);` returns a pointer with the value 0x00000002 for instance which doesn't make any sens but enough to pass the following condition, and finally deferencing the pointer causes a crash.

Then I saw that the malloc mutex wasn't initialized and executing a `os_mutex_init` before the first `os_malloc` solved my issue.

My crash happens in `os_mutex_release`, but the behavior in `os_mutex_pend` is also undefined, in function of the value present in the mutex structure during its declaration.

I hope it helps you to understand.

clmklk, 2019-06-21:
Sorry I realize than I did not mentioned the mutex name: `os_malloc_mutex`.

ccollins476ad, 2019-06-21:
`os_malloc_mutex` is defined as (`os_heap.c`):
```
#if MYNEWT_VAL(OS_SCHEDULING)
static struct os_mutex os_malloc_mutex;
#endif
```

So this object is in bss.  It should get zero-initialized at startup.

You are seeing that this object contains garbage values on first use?

utzig, 2019-06-21:
Also BSS is manually zeroed in the startup_*.s file so if you are working on a new port this might be missing...

clmklk, 2019-06-24:
Thank you for the answers.
I really missed the part with the bss section.
Yes indeed I am have a structure not fully zeroed but I am working on our port of the STM32L0 which seems to have some differences in the startup and linker file. My issue may come from there.

Now I understand how you implemented this part and how it should behave, so I close the issue.

closed 2019-06-24 by clmklk

clmklk, 2019-07-31:
For information, we found the root cause of the issue.
The BSS section was set properly to 0, but the value at a specific address was changing causing troubles after that.
The issue was due to interrupts firing (Systick for instance) while jumping to the application. The counter measure we took, is to disable all interrupts in the bootloader, just before jumping.