apache/mynewt-core #1925: sys/config test failed on Windows

damageg, 2019-07-22:
Following the install guide on Windows 10 (MSYS2 64 bit),  I created a first project and ran

newt test @apache-mynewt-core/sys/config

I got:
repos/apache-mynewt-core/hw/mcu/native/src/hal_uart.c:281:9: error: implicit declaration of function ‘openpty’; did you mean ‘openat’? [-Werror=implicit-function-declaration]
  281 |     if (openpty(&fd, &loop_slave, pty_name, NULL, NULL) < 0) {
      |         ^~~~~~~
      |         openat
cc1: all warnings being treated as errors

damageg, 2019-07-22:
It's not supported on Windows - Doh.

closed 2019-07-22 by damageg

rajithavk, 2021-05-30:
Does it mean, mynewt is not supported for windows?
because the blinky starter project also fails with the same error.