apache/mynewt-core #1948: Error: Could not resolve app package: @apache-mynewt-core/apps/boot

OTLabs, 2019-08-06:
I use latest `newt` 1.7.0 on Windows to build `rbnano2_boot` target and get follwoing error:

```
$ newt version
Apache Newt
   Version: 1.7.0
  Git Hash: 80bcba7
Build Date: 2019-07-17_13:51

$ newt build rbnano2_boot
Building target targets/rbnano2_boot
Error: Could not resolve app package: @apache-mynewt-core/apps/boot
```

utzig, 2019-08-06:
If you have a recent tree of Mynewt you have to change to boot app, using something like: `newt target set rbnano2_boot app=@mcuboot/boot/mynewt`.

OTLabs, 2019-08-07:
Implementing this setting I got the build running and successfully build the target, thank you!

I was following step-by-step the Tutorial | Project Blinky | Blinky on RedBear Nano 2 (version: latest, https://mynewt.apache.org/latest/tutorials/blinky/rbnano2.html). I would quite appreciate if the latest documentation can reflect this change in settings.

utzig, 2019-08-07:
Done here: https://github.com/apache/mynewt-documentation/pull/76

OTLabs, 2019-08-07:
Thanks a lot!

utzig, 2019-08-09:
This was fixed with apache/mynewt-documentation#76

closed 2019-08-09 by utzig

brainstorm, 2019-11-01:
I'm facing this same issue with this repo:

https://github.com/chris-zen/mdp-link-mynewt

And mynewt HEAD installed with Homebrew:

```shell
$ newt build nrf52_boot
Downloading repository mynewt-core (commit: master) from https://github.com/apache/mynewt-core.git
Downloading repository mynewt_nrf52840_mdk (commit: master) from https://github.com/makerdiary/mynewt_nrf52840_mdk.git
Downloading repository tinyusb (commit: master) from https://github.com/hathach/tinyusb.git
Building target targets/nrf52_boot
Error: Could not resolve app package: @mcuboot/boot/mynewt
```

Also with the arm-eabi toolchain outlined on issue #2076, from ArmMbed. I tried following the updated docs, using `set` for the bst and the `optimized` compile profile, but I get the same effect.

Are you able to reproduce this issue, @utzig?

utzig, 2019-11-01:
@brainstorm Yes, I could reproduce. There's many things that could be going wrong here. But doing `newt upgrade -f` afterwards solved it (not that it should be happening in the first place!). Also a very recent build of `newt` is required, because of a known bug in `git` submodules that `newt` is currently "masking" for the mcuboot repo.

Thilanka97, 2020-02-12:
hey @brainstorm , @utzig I am facing the same problem with blinky project.
 _**Could not resolve app package: @mcuboot/boot/mynewt**_

 I tried _**newt upgrade -f**_ as you suggested but I am getting following error with that cmd. 
_**Error: usage: git rev-list [OPTION] <commit-id>... [ -- paths... ]**_

What is wrong here. I have already installed git as well.  Please help me to solve this problem. 

Thanks in advance!

brainstorm, 2020-02-23:
@Thilanka97 I'm not sure if it's entirely related, but there are some ongoing efforts to fix those issues in https://github.com/apache/mynewt-core/issues/2125... I would come back when 2.0 is properly merged/tested/released, IMHO.