apache/mynewt-core #1949: newt target show shows too much info:

OTLabs, 2019-08-06:
Latest `newt target show` on Windows shows too much info for `rbnano2 blinky` project:

```
$ newt target show
@mcuboot/targets/basic
    app=@mcuboot/boot/mynewt
    bsp=@apache-mynewt-core/hw/bsp/nordic_pca10056
    build_profile=optimized
    syscfg=BOOTUTIL_BOOTSTRAP=0:BOOTUTIL_ENCRYPT_KW=0:BOOTUTIL_ENCRYPT_RSA=0:BOOTUTIL_HAVE_LOGGING=0:BOOTUTIL_LOG_LEVEL=BOOTUTIL_LOG_LEVEL_INFO:BOOTUTIL_MAX_IMG_SECTORS=256:BOOTUTIL_NO_LOGGING=1:BOOTUTIL_OVERWRITE_ONLY=0:BOOTUTIL_OVERWRITE_ONLY_FAST=1:BOOTUTIL_SIGN_EC256=0:BOOTUTIL_SIGN_RSA=0:BOOTUTIL_USE_MBED_TLS=0:BOOTUTIL_USE_TINYCRYPT=1:BOOTUTIL_VALIDATE_SLOT0=0:BOOT_SERIAL=0:BOOT_SERIAL_DETECT_PIN=11:BOOT_SERIAL_DETECT_PIN_VAL=0:BOOT_SERIAL_REPORT_PIN=13:CONSOLE_COMPAT=1:CONSOLE_INPUT=0:CONSOLE_RTT=0:CONSOLE_UART=0:MBEDTLS_NIST_KW_C=0:OS_CPUTIME_TIMER_NUM=0:TIMER_0=1:UART_0=0
@mcuboot/targets/bootserial
    app=@mcuboot/boot/mynewt
    bsp=@apache-mynewt-core/hw/bsp/nordic_pca10056
    build_profile=optimized
    syscfg=BOOTUTIL_BOOTSTRAP=0:BOOTUTIL_ENCRYPT_KW=0:BOOTUTIL_ENCRYPT_RSA=0:BOOTUTIL_HAVE_LOGGING=0:BOOTUTIL_LOG_LEVEL=BOOTUTIL_LOG_LEVEL_INFO:BOOTUTIL_MAX_IMG_SECTORS=256:BOOTUTIL_NO_LOGGING=1:BOOTUTIL_OVERWRITE_ONLY=0:BOOTUTIL_OVERWRITE_ONLY_FAST=1:BOOTUTIL_SIGN_EC256=0:BOOTUTIL_SIGN_RSA=0:BOOTUTIL_USE_MBED_TLS=0:BOOTUTIL_USE_TINYCRYPT=1:BOOTUTIL_VALIDATE_SLOT0=0:BOOT_SERIAL=1:BOOT_SERIAL_DETECT_PIN=11:BOOT_SERIAL_DETECT_PIN_VAL=0:BOOT_SERIAL_REPORT_PIN=13:CONSOLE_COMPAT=1:CONSOLE_INPUT=0:CONSOLE_RTT=0:CONSOLE_UART=0:MBEDTLS_NIST_KW_C=0:OS_CPUTIME_TIMER_NUM=0:TIMER_0=1:UART_0=1
@mcuboot/targets/ecdsa
    app=@mcuboot/boot/mynewt
    bsp=@apache-mynewt-core/hw/bsp/nordic_pca10056
    build_profile=optimized
    key_file=@mcuboot/root-ec-p256.pem
    syscfg=BOOTUTIL_BOOTSTRAP=0:BOOTUTIL_ENCRYPT_KW=0:BOOTUTIL_ENCRYPT_RSA=0:BOOTUTIL_HAVE_LOGGING=0:BOOTUTIL_LOG_LEVEL=BOOTUTIL_LOG_LEVEL_INFO:BOOTUTIL_MAX_IMG_SECTORS=256:BOOTUTIL_NO_LOGGING=1:BOOTUTIL_OVERWRITE_ONLY=0:BOOTUTIL_OVERWRITE_ONLY_FAST=1:BOOTUTIL_SIGN_EC256=1:BOOTUTIL_SIGN_RSA=0:BOOTUTIL_USE_MBED_TLS=0:BOOTUTIL_USE_TINYCRYPT=1:BOOTUTIL_VALIDATE_SLOT0=1:BOOT_SERIAL=0:BOOT_SERIAL_DETECT_PIN=11:BOOT_SERIAL_DETECT_PIN_VAL=0:BOOT_SERIAL_REPORT_PIN=13:CONSOLE_COMPAT=1:CONSOLE_INPUT=0:CONSOLE_RTT=0:CONSOLE_UART=0:MBEDTLS_NIST_KW_C=0:OS_CPUTIME_TIMER_NUM=0:TIMER_0=1:UART_0=0
@mcuboot/targets/ecdsa_kw
    app=@mcuboot/boot/mynewt
    bsp=@apache-mynewt-core/hw/bsp/nordic_pca10056
    build_profile=optimized
    key_file=@mcuboot/root-ec-p256.pem
    syscfg=BOOTUTIL_BOOTSTRAP=0:BOOTUTIL_ENCRYPT_KW=1:BOOTUTIL_ENCRYPT_RSA=0:BOOTUTIL_HAVE_LOGGING=0:BOOTUTIL_LOG_LEVEL=BOOTUTIL_LOG_LEVEL_INFO:BOOTUTIL_MAX_IMG_SECTORS=256:BOOTUTIL_NO_LOGGING=1:BOOTUTIL_OVERWRITE_ONLY=0:BOOTUTIL_OVERWRITE_ONLY_FAST=1:BOOTUTIL_SIGN_EC256=1:BOOTUTIL_SIGN_RSA=0:BOOTUTIL_USE_MBED_TLS=0:BOOTUTIL_USE_TINYCRYPT=1:BOOTUTIL_VALIDATE_SLOT0=1:BOOT_SERIAL=0:BOOT_SERIAL_DETECT_PIN=11:BOOT_SERIAL_DETECT_PIN_VAL=0:BOOT_SERIAL_REPORT_PIN=13:CONSOLE_COMPAT=1:CONSOLE_INPUT=0:CONSOLE_RTT=0:CONSOLE_UART=0:MBEDTLS_NIST_KW_C=0:OS_CPUTIME_TIMER_NUM=0:TIMER_0=1:UART_0=0
@mcuboot/targets/rsa
    app=@mcuboot/boot/mynewt
    bsp=@apache-mynewt-core/hw/bsp/nordic_pca10056
    build_profile=optimized
    key_file=@mcuboot/root-rsa-2048.pem
    syscfg=BOOTUTIL_BOOTSTRAP=0:BOOTUTIL_ENCRYPT_KW=0:BOOTUTIL_ENCRYPT_RSA=0:BOOTUTIL_HAVE_LOGGING=0:BOOTUTIL_LOG_LEVEL=BOOTUTIL_LOG_LEVEL_INFO:BOOTUTIL_MAX_IMG_SECTORS=256:BOOTUTIL_NO_LOGGING=1:BOOTUTIL_OVERWRITE_ONLY=0:BOOTUTIL_OVERWRITE_ONLY_FAST=1:BOOTUTIL_SIGN_EC256=0:BOOTUTIL_SIGN_RSA=1:BOOTUTIL_USE_MBED_TLS=1:BOOTUTIL_USE_TINYCRYPT=0:BOOTUTIL_VALIDATE_SLOT0=1:BOOT_SERIAL=0:BOOT_SERIAL_DETECT_PIN=11:BOOT_SERIAL_DETECT_PIN_VAL=0:BOOT_SERIAL_REPORT_PIN=13:CONSOLE_COMPAT=1:CONSOLE_INPUT=0:CONSOLE_RTT=0:CONSOLE_UART=0:MBEDTLS_NIST_KW_C=0:OS_CPUTIME_TIMER_NUM=0:TIMER_0=1:UART_0=0
@mcuboot/targets/rsa_kw
    app=@mcuboot/boot/mynewt
    bsp=@apache-mynewt-core/hw/bsp/nordic_pca10056
    build_profile=optimized
    key_file=@mcuboot/root-rsa-2048.pem
    syscfg=BOOTUTIL_BOOTSTRAP=0:BOOTUTIL_ENCRYPT_KW=1:BOOTUTIL_ENCRYPT_RSA=0:BOOTUTIL_HAVE_LOGGING=0:BOOTUTIL_LOG_LEVEL=BOOTUTIL_LOG_LEVEL_INFO:BOOTUTIL_MAX_IMG_SECTORS=256:BOOTUTIL_NO_LOGGING=1:BOOTUTIL_OVERWRITE_ONLY=0:BOOTUTIL_OVERWRITE_ONLY_FAST=1:BOOTUTIL_SIGN_EC256=0:BOOTUTIL_SIGN_RSA=1:BOOTUTIL_USE_MBED_TLS=1:BOOTUTIL_USE_TINYCRYPT=0:BOOTUTIL_VALIDATE_SLOT0=1:BOOT_SERIAL=0:BOOT_SERIAL_DETECT_PIN=11:BOOT_SERIAL_DETECT_PIN_VAL=0:BOOT_SERIAL_REPORT_PIN=13:CONSOLE_COMPAT=1:CONSOLE_INPUT=0:CONSOLE_RTT=0:CONSOLE_UART=0:MBEDTLS_NIST_KW_C=1:OS_CPUTIME_TIMER_NUM=0:TIMER_0=1:UART_0=0
@mcuboot/targets/rsa_overwriteonly
    app=@mcuboot/boot/mynewt
    bsp=@apache-mynewt-core/hw/bsp/nordic_pca10056
    build_profile=optimized
    key_file=@mcuboot/root-rsa-2048.pem
    syscfg=BOOTUTIL_BOOTSTRAP=0:BOOTUTIL_ENCRYPT_KW=0:BOOTUTIL_ENCRYPT_RSA=0:BOOTUTIL_HAVE_LOGGING=0:BOOTUTIL_LOG_LEVEL=BOOTUTIL_LOG_LEVEL_INFO:BOOTUTIL_MAX_IMG_SECTORS=256:BOOTUTIL_NO_LOGGING=1:BOOTUTIL_OVERWRITE_ONLY=1:BOOTUTIL_OVERWRITE_ONLY_FAST=1:BOOTUTIL_SIGN_EC256=0:BOOTUTIL_SIGN_RSA=1:BOOTUTIL_USE_MBED_TLS=1:BOOTUTIL_USE_TINYCRYPT=0:BOOTUTIL_VALIDATE_SLOT0=1:BOOT_SERIAL=0:BOOT_SERIAL_DETECT_PIN=11:BOOT_SERIAL_DETECT_PIN_VAL=0:BOOT_SERIAL_REPORT_PIN=13:CONSOLE_COMPAT=1:CONSOLE_INPUT=0:CONSOLE_RTT=0:CONSOLE_UART=0:MBEDTLS_NIST_KW_C=0:OS_CPUTIME_TIMER_NUM=0:TIMER_0=1:UART_0=0
@mcuboot/targets/rsa_rsaoaep
    app=@mcuboot/boot/mynewt
    bsp=@apache-mynewt-core/hw/bsp/nordic_pca10056
    build_profile=optimized
    key_file=@mcuboot/root-rsa-2048.pem
    syscfg=BOOTUTIL_BOOTSTRAP=0:BOOTUTIL_ENCRYPT_KW=0:BOOTUTIL_ENCRYPT_RSA=1:BOOTUTIL_HAVE_LOGGING=0:BOOTUTIL_LOG_LEVEL=BOOTUTIL_LOG_LEVEL_INFO:BOOTUTIL_MAX_IMG_SECTORS=256:BOOTUTIL_NO_LOGGING=1:BOOTUTIL_OVERWRITE_ONLY=0:BOOTUTIL_OVERWRITE_ONLY_FAST=1:BOOTUTIL_SIGN_EC256=0:BOOTUTIL_SIGN_RSA=1:BOOTUTIL_USE_MBED_TLS=1:BOOTUTIL_USE_TINYCRYPT=0:BOOTUTIL_VALIDATE_SLOT0=1:BOOT_SERIAL=0:BOOT_SERIAL_DETECT_PIN=11:BOOT_SERIAL_DETECT_PIN_VAL=0:BOOT_SERIAL_REPORT_PIN=13:CONSOLE_COMPAT=1:CONSOLE_INPUT=0:CONSOLE_RTT=0:CONSOLE_UART=0:MBEDTLS_NIST_KW_C=0:OS_CPUTIME_TIMER_NUM=0:TIMER_0=1:UART_0=0
@mcuboot/targets/rsa_rsaoaep_bootstrap
    app=@mcuboot/boot/mynewt
    bsp=@apache-mynewt-core/hw/bsp/nordic_pca10056
    build_profile=optimized
    key_file=@mcuboot/root-rsa-2048.pem
    syscfg=BOOTUTIL_BOOTSTRAP=1:BOOTUTIL_ENCRYPT_KW=0:BOOTUTIL_ENCRYPT_RSA=1:BOOTUTIL_HAVE_LOGGING=0:BOOTUTIL_LOG_LEVEL=BOOTUTIL_LOG_LEVEL_INFO:BOOTUTIL_MAX_IMG_SECTORS=256:BOOTUTIL_NO_LOGGING=1:BOOTUTIL_OVERWRITE_ONLY=0:BOOTUTIL_OVERWRITE_ONLY_FAST=1:BOOTUTIL_SIGN_EC256=0:BOOTUTIL_SIGN_RSA=1:BOOTUTIL_USE_MBED_TLS=1:BOOTUTIL_USE_TINYCRYPT=0:BOOTUTIL_VALIDATE_SLOT0=1:BOOT_SERIAL=0:BOOT_SERIAL_DETECT_PIN=11:BOOT_SERIAL_DETECT_PIN_VAL=0:BOOT_SERIAL_REPORT_PIN=13:CONSOLE_COMPAT=1:CONSOLE_INPUT=0:CONSOLE_RTT=0:CONSOLE_UART=0:MBEDTLS_NIST_KW_C=0:OS_CPUTIME_TIMER_NUM=0:TIMER_0=1:UART_0=0
targets/my_blinky_sim
    app=apps/blinky
    bsp=@apache-mynewt-core/hw/bsp/native
    build_profile=debug
targets/rbnano2_blinky
    app=apps/blinky
    bsp=@apache-mynewt-core/hw/bsp/rb-nano2
    build_profile=debug
targets/rbnano2_boot
    app=@apache-mynewt-core/apps/boot
    bsp=@apache-mynewt-core/hw/bsp/rb-nano2
    build_profile=optimized
```

utzig, 2019-08-06:
You are running `newt target show` without passing it the target you want to show, so it will just show all the ones it finds in the tree, you might use `newt target show rbnano2 blinky` instead.

OTLabs, 2019-08-07:
`newt target show rbnano2_blinky` worked like a charm! Thank you!

I was following step-by-step the Tutorial | Project Blinky | Blinky on RedBear Nano 2 (version: latest, https://mynewt.apache.org/latest/tutorials/blinky/rbnano2.html). I would quite appreciate if the latest documentation can reflect this change in settings.

closed 2019-08-09 by utzig