apache/mynewt-core #1994: Test fails to execute on Mac OS (Mojave)

Manish-Juul, 2019-09-11:
OS: macOS Mojave
Version: 10.14.6

Following is the output as I execute the test command on console.

$ newt test @apache-mynewt-core/sys/config
Testing package @apache-mynewt-core/sys/config/selftest-fcb
Compiling repos/apache-mynewt-core/crypto/tinycrypt/src/cbc_mode.c
Compiling repos/apache-mynewt-core/crypto/tinycrypt/src/ecc_dsa.c
. . .
. . .
Compiling repos/apache-mynewt-core/util/rwlock/src/rwlock.c
Compiling repos/apache-mynewt-core/util/mem/src/mem.c
Archiving crypto_tinycrypt.a
Archiving encoding_base64.a
. . .
. . .
Archiving util_mem.a
Archiving util_rwlock.a
Linking /Users/manish/workspace/mynewt-newt/newt/my_prj/bin/targets/unittest/sys_config_selftest-fcb/app/sys/config/selftest-fcb/sys_config_selftest-fcb.elf
**ld: warning: The i386 architecture is deprecated for macOS (remove from the Xcode build setting: ARCHS)**
ld: warning: ignoring file /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/lib/libm.tbd, missing required architecture i386 in file /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/lib/libm.tbd
ld: warning: ignoring file /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/lib/libSystem.tbd, missing required architecture i386 in file /Library/Developer/CommandLineTools/SDKs/MacOSX10.14.sdk/usr/lib/libSystem.tbd
Undefined symbols for architecture i386:
  "__Exit", referenced from:
      ___assert_func in kernel_os.a(os_arch.o)
  "___assert_rtn", referenced from:
      _flash_area_id_from_image_slot in sys_flash_map.a(flash_map.o)
      _flash_map_init in sys_flash_map.a(flash_map.o)
      _tu_case_fail_assert in test_testutil.a(case.o)
      _tu_case_append_file_info in test_testutil.a(case.o)
      _tu_case_append_assert_msg in test_testutil.a(case.o)
      _tu_case_pass_manual in test_testutil.a(case.o)
      _tu_case_append_manual_pass_msg in test_testutil.a(case.o)
      ...

chris-zen, 2019-10-13:
@manish-sudo you are probably using the MacOS default gcc, which uses clang. You can follow the documentation to install other versions of gcc.

But this same test also fails for me even when using the recommended versions in the documentation (tried with gcc 5, 6 and 9 getting different kind of errors):
```
Compiling repos/apache-mynewt-core/crypto/tinycrypt/src/ecc_dsa.c
In file included from /usr/local/Cellar/gcc@5/5.5.0_3/lib/gcc/5/gcc/x86_64-apple-darwin17.7.0/5.5.0/include/stdint.h:9:0,
                 from repos/apache-mynewt-core/crypto/tinycrypt/include/tinycrypt/aes.h:52,
                 from repos/apache-mynewt-core/crypto/tinycrypt/src/aes_decrypt.c:33:
/usr/local/Cellar/gcc@5/5.5.0_3/lib/gcc/5/gcc/x86_64-apple-darwin17.7.0/5.5.0/include-fixed/stdint.h:27:32: fatal error: sys/_types/_int8_t.h: No such file or directory
compilation terminated.
Error: Test failure(s):
Passed tests: []
Failed tests: [sys/config/selftest-nffs sys/config/selftest-fcb]
```

rquitnik, 2019-11-14:
Same issue as @chris-zen: I am getting the same errors with gcc5, 6, 8, 9.

Error: Test failure(s):
Passed tests: []
Failed tests: [sys/config/selftest-fcb sys/config/selftest-nffs]

I'm using gcc8 in this case (as defined in compiler.yml), but I get a number of MacOSX related errors, also:

In file included from /usr/local/Cellar/gcc@8/8.3.0/lib/gcc/8/gcc/x86_64-apple-darwin18/8.3.0/include-fixed/stdint.h:27,
                 from /usr/local/Cellar/gcc@8/8.3.0/lib/gcc/8/gcc/x86_64-apple-darwin18/8.3.0/include/stdint.h:9,
                 from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/net/if_var.h:68,
                 from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/net/if.h:75,
                 from repos/apache-mynewt-core/net/ip/native_sockets/src/native_itf.c:20:

In file included from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/machine/_types.h:32,
                 from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/sys/_types.h:33,
                 from /usr/local/Cellar/gcc@8/8.3.0/lib/gcc/8/gcc/x86_64-apple-darwin18/8.3.0/include-fixed/stdint.h:61,
                 from /usr/local/Cellar/gcc@8/8.3.0/lib/gcc/8/gcc/x86_64-apple-darwin18/8.3.0/include/stdint.h:9,
                 from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/net/if_var.h:68,
                 from /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/net/if.h:75,
                 from repos/apache-mynewt-core/net/ip/native_sockets/src/native_itf.c:20: