apache/mynewt-core #2033: newt upgrade  failed

Huihh, 2019-10-10:
Hi, Engineer:
Occur below error when I setup mynewt environment with <newt upgrade> command.
Host: unbuntu 18.04
what should I do, Please help me, thanks very mush! 

xxx@xxx:~/my_workspace/myproj$ newt upgrade
Downloading repository incubator-mynewt-core (commit: master) from https://github.com/apache/incubator-mynewt-core.git
Error: Error downloading repository apache-mynewt-core: Cloning into '/tmp/newt-repo347787386'...
error: RPC failed; curl 56 GnuTLS recv error (-54): Error in the pull function.
fatal: The remote end hung up unexpectedly
fatal: early EOF
fatal: index-pack failed

ccollins476ad, 2019-10-10:
Thanks @Huihh.  I have seen something similar in the past.  In my case, a security program running on my computer (zscaler) was automatically terminating some connections to github.  When I quit zscaler, the problem went away.

Are you running any firewall-esque programs like zscaler?

Huihh, 2019-10-12:
Hi, @ccollins476ad. I want to build mynewt environment on unbutu 19.04, I tried follow "Quick Start" on home web, but failed, I want to used mynewt to my IOT products, but now, I can`t build environment on ubuntu 19.04, Please give me some suggest, Thanks very much

KKopyscinski, 2021-03-10:
I don't think there's any issue with newt upgrade, I tested it again on Ubuntu 20.04 and Newt `Apache Newt 1.9.0-dev / 5da6e2f / 2021-02-11_13:34`. I will close it for now, please reopen if problem persists

closed 2021-03-10 by KKopyscinski