apache/mynewt-core #2040: Build failure due to implicit declaration of 'static_assert'

mehulhirpara, 2019-10-15:
Unable to build blemesh_light target on latest code (0-dev).

**Compilation Error**
`
Error: repos/apache-mynewt-core/kernel/os/src/os_mempool.c: In function 'os_memblock_from':
repos/apache-mynewt-core/kernel/os/src/os_mempool.c:308:5: error: implicit declaration of function 'static_assert' [-Werror=implicit-function-declaration]
     static_assert(sizeof block_addr == sizeof baddr32,
     ^
cc1: all warnings being treated as errors
`

**Target YML:**
`
target.app: "@apache-mynewt-nimble/apps/blemesh_light"
target.bsp: "@apache-mynewt-core/hw/bsp/nordic_pca10056"
target.build_profile: debug
`

**Syscfg YML:**
`
syscfg.vals:
    BLE_MESH_DEV_UUID: "(uint8_t[16]){0x22, 0x20, 0}"
    BLE_MESH_PB_GATT: 1
`

**Package YML:**
`
pkg.name: targets/blemesh_light
pkg.type: target
pkg.description: 
pkg.author: 
pkg.homepage: 
`

andrzej-kaczmarek, 2019-10-15:
which compiler/version do you use? this builds fine with GCC

mehulhirpara, 2019-10-15:
newt version
Apache Newt
   Version: 1.7.0
  Git Hash: unknown
Build Date: 2019-09-03_13:44

mehulhirpara, 2019-10-23:
update:
with docker based newt, I am able to compile.

deepthits, 2019-10-29:
What is the solution for this issue? Even I am facing this

andrzej-kaczmarek, 2019-10-29:
imo we should change `static_assert` (C++ symbol) to `_Static_assert` (C11 symbol) - we anyway have `#define` to redefine the former to the latter. not sure if this will be supported by old GCC, but works fine in 7.x+.

kasjer, 2019-11-15:
I would opt to change assert.h not to check compiler version when static_assert is defined.
It is mentioned in cppreference that convenience macro is defined for _Static_assert starting from C11 but it may be visible on gcc without C11.
Then if _Static_assert keyword is not present compilation will fail anyway.
```
#if !defined __cplusplus
#define static_assert _Static_assert
#endif
```

PR #2417 (2020-11-24): baselibc: Always define static_assert for C
Currently **static_assert** is only defined when
```__STDC_VERSION__ >= 201112L```.

Static asserts present in header files should
use static_assert to be able to build with C++.

For non C++ build it should map **_Static_assert**.
While _Static_assert is C11 feature it can be
present for older compilers.

This drops checking for compiler version,
if compiler does not support _Static_assert it
will fail anyway.

This resolves #2040
closed 2020-11-24 by kasjer