apache/mynewt-core #2076: multiple definition of `fflush'

brainstorm, 2019-10-30:
While compiling this mynewt project: https://github.com/chris-zen/mdp-link-mynewt/ using the HEAD version of mynewt:

```shell
$ newt version
Apache Newt
   Version: 1.8.0-dev
  Git Hash: d703a10-dirty
Build Date: 2019-10-30_23:28
```

Running OSX Catalina 10.15, but the system's compiler is not cross-compiling this anyway (see further below for the affected GCC)

```
$ gcc --version
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Apple clang version 11.0.0 (clang-1100.0.33.8)
Target: x86_64-apple-darwin19.0.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

Anyway, the core of the issue is that I'm faced with this:

```shell
(...)
Archiving @tinyusb_tinyusb.a
Archiving @apache-mynewt-core_util_mem.a
Archiving @apache-mynewt-core_util_rwlock.a
Linking /Users/romanvg/dev/mdp/mdp-link-mynewt/bin/targets/nrf52_mdp_link/app/apps/mdp-link/mdp-link.elf
Error: /usr/local/Cellar/arm-none-eabi-gcc/20160926/bin/../lib/gcc/arm-none-eabi/5.4.1/../../../../arm-none-eabi/lib/armv7e-m/libg.a(lib_a-fflush.o): In function `fflush':
fflush.c:(.text.fflush+0x0): multiple definition of `fflush'
/Users/romanvg/dev/mdp/mdp-link-mynewt/bin/targets/nrf52_mdp_link/app/@apache-mynewt-core/libc/baselibc/@apache-mynewt-core_libc_baselibc.a(inline.o):/Users/romanvg/dev/mdp/mdp-link-mynewt/repos/apache-mynewt-core/libc/baselibc/include/stdio.h:99: first defined here
collect2: error: ld returned 1 exit status
```

Which I temporarily solve modifying the `stdio.h` header under `repos/apache-mynewt-core/libc/baselibc/include/stdio.h`:

```diff
$ git diff stdio.h.orig stdio.h
diff --git a/libc/baselibc/include/stdio.h b/libc/baselibc/include/stdio.h
index fa7103fd0..8ec0d27e1 100644
--- a/libc/baselibc/include/stdio.h
+++ b/libc/baselibc/include/stdio.h
@@ -93,10 +93,12 @@ __extern_inline char *strerror(int errnum)
 #define getc(f) fgetc(f)
 #define getchar() fgetc(stdin)

+/*
 __extern_inline int fflush(FILE *stream)
 {
        return 0;
 }
+*/

 __extern int printf(const char *, ...);
 __extern int vprintf(const char *, va_list);
```

And then it compiles/links successfully:

```
$ newt build nrf52_mdp_link
Building target targets/nrf52_mdp_link
Target successfully built: targets/nrf52_mdp_link
```

I guess I should get rid of my relatively old gcc homebrew dependency with proxmark3?:

```
$ brew info arm-none-eabi-gcc
proxmark/proxmark3/arm-none-eabi-gcc: stable 20160926
GCC for embedded ARM processors
https://launchpad.net/gcc-arm-embedded
/usr/local/Cellar/arm-none-eabi-gcc/20160926 (5,502 files, 447.0MB) *
  Built from source on 2017-12-09 at 00:02:56
From: https://github.com/proxmark/homebrew-proxmark3/blob/master/arm-none-eabi-gcc.rb
```

Any suggestions on which brew formula to pull a newer compiler from are welcome since there seems to only be one `arm-none-eabi-gcc` on my Homebrew:

```shell
$ brew search arm-none-eabi
==> Formulae
proxmark/proxmark3/arm-none-eabi-gcc
```

brainstorm, 2019-11-01:
I went for this Homebrew toolchain instead which seems to be more recent and better supported, with ties to Rust embedded ecosystem:

```shell
$ brew tap ArmMbed/homebrew-formulae
$ brew install arm-none-eabi-gcc
```

mkiiskila, 2019-11-01:
I'm guessing that your app references something in libc, and that particular function (or var) is not present in baselibc. As a result, you're also linking against newlib. And it's bringing in object file from newlib .a which includes it's version of fflush().

There's 2 things to do here: 1st figure out what brings in newlib, and then 2nd either get rid of that call, or implement a version of that within baselibc.

To do 1, you could take a look at, e.g. of the .map file which gets generated to same directory as the elf file. File contains the symbol names included in the linked result, and shows which .a that symbol came from. Basically look for that libg.a you can see in the error message. In a nicely contained build, you should see nothing being brought in. You can check the .lst file (again in the same directory) to see who calls that routine.

brainstorm, 2019-11-01:
Thanks @mkiiskila for the reply, here's a snippet of my generated `.map` file:

```shell
$ grep fflush ./bin/targets/nrf52_mdp_link/app/apps/mdp-link/mdp-link.elf.map
/usr/local/Cellar/arm-none-eabi-gcc/8-2018-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg.a(lib_a-fflush.o)
                              /usr/local/Cellar/arm-none-eabi-gcc/8-2018-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg.a(lib_a-wbuf.o) (_fflush_r)
 .text.fflush   0x0000000000000000        0x4 /Users/romanvg/dev/mdp/mdp-link-mynewt/bin/targets/nrf52_mdp_link/app/@apache-mynewt-core/libc/baselibc/@apache-mynewt-core_libc_baselibc.a(inline.o)
 .text          0x0000000000000000        0x0 /usr/local/Cellar/arm-none-eabi-gcc/8-2018-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg.a(lib_a-fflush.o)
 .data          0x0000000000000000        0x0 /usr/local/Cellar/arm-none-eabi-gcc/8-2018-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg.a(lib_a-fflush.o)
 .bss           0x0000000000000000        0x0 /usr/local/Cellar/arm-none-eabi-gcc/8-2018-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg.a(lib_a-fflush.o)
 .text.fflush   0x0000000000000000       0x24 /usr/local/Cellar/arm-none-eabi-gcc/8-2018-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg.a(lib_a-fflush.o)
                0x000000000001198c      0x14c /usr/local/Cellar/arm-none-eabi-gcc/8-2018-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg.a(lib_a-fflush.o)
 .text._fflush_r
                0x0000000000011ad8       0x5c /usr/local/Cellar/arm-none-eabi-gcc/8-2018-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg.a(lib_a-fflush.o)
                0x0000000000011ad8                _fflush_r
                0x0000000000000a89       0x2e /usr/local/Cellar/arm-none-eabi-gcc/8-2018-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg.a(lib_a-fflush.o)
 .debug_frame   0x0000000000003090       0x5c /usr/local/Cellar/arm-none-eabi-gcc/8-2018-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg.a(lib_a-fflush.o)
```

brainstorm, 2019-11-01:
There's also no trace of newlib on my build:

```
$ find . -iname "newlib*"
$
```

brainstorm, 2019-11-01:
Oh, btw, I'm running on OSX 10.15 Catalina, which dropped support for 32 bits, could that be the issue?

mkiiskila, 2019-11-01:
Do 'grep libg.a ./bin/targets/nrf52_mdp_link/app/apps/mdp-link/mdp-link.elf.map', rather.
It's not a call to fflush() which brings in newlib, it's something else. Something which is not implemented in baselibc.

Compiling on Catalina is ok; you'd only have trouble compiling code to run on MacOS itself (i.e. unittests or simulator).

mkiiskila, 2019-11-01:
The crosscompiling gcc you have installed comes with newlib included (all libc.a & libg.a under /usr/local/Cellar/arm-none-eabi-gcc/8-2018-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/lib/).
We don't specifically stop people from bringing in things from there, as they might really, really need something from there. That's why we allow the linking. Downside is that you're bringing a lot of things you don't actually want, along with the thing you need.

You can also look at the manifest.json file (same place as the .map and .lst), it lists all the symbols which came from libg.a. I'm guessing you got at least another copy of malloc (which will not ever get called).

Happy hunting!

brainstorm, 2019-11-01:
I cannot see what is pulling newlib, TBH, all seems to come from the arm-eabi toolchain itself :-S

[grep_libg.a.map.txt](https://github.com/apache/mynewt-core/files/3797065/lib.a.map.txt)

Here's the full `.map` file:

[full_mdp-link.elf.map.txt](https://github.com/apache/mynewt-core/files/3797071/mdp-link.elf.map.txt)

brainstorm, 2019-11-01:
Hm, I see what you mean. Is there a way to selectively blacklist symbols from being linked from elsewhere? There's no `manifest.json` file either :/

mkiiskila, 2019-11-01:
I did take a look, and I'm guessing the offender are calls to putchar() from tinyusb library. Are you interested in implementing putchar() for baselibc?

mkiiskila, 2019-11-01:
these 2 lines from your .map:
/usr/local/Cellar/arm-none-eabi-gcc/8-2018-q4-major/gcc/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg.a(lib_a-putchar.o)
                              /Users/romanvg/dev/mdp/mdp-link-mynewt/bin/targets/nrf52_mdp_link/app/@tinyusb/tinyusb/@tinyusb_tinyusb.a(tusb.o) (putchar)

brainstorm, 2019-11-01:
I'm not familiar with baselibc, but seems to be implemented in here?:

https://github.com/apache/mynewt-core/blob/master/libc/baselibc/include/stdio.h#L92

OTOH, how come that commenting the `fflush` inline function on https://github.com/apache/mynewt-core/blob/master/libc/baselibc/include/stdio.h#L96, solves the issue and compilation succeeds? :-S

Other baselibc implementations do not seem to have that fflush there:

https://github.com/PetteriAimonen/Baselibc/blob/master/include/stdio.h

mkiiskila, 2019-11-01:
Indeed, just saw that we do have an implementation in baselibc. Not sure why the libg implementation is preferred. Maybe one of files including calls to putchar() is missing #include <stdio.h>? You'll have to read the .lst file (or do objdump -S of your .elf) to see where the call to putchar is located.

fflush() does not have any meaning for us. We don't have extra buffering in fwrite() level.

brainstorm, 2019-11-01:
Would it make sense then to just remove that `__extern_inline fflush` then?

Happy to pullrequest that simple change if you reckon there is no breakage for other components/apps down the line.

brainstorm, 2019-11-01:
After removing that fflush from stdio, here's the `.lst` (if I don't remove it, no .lst gets generated): [mdp-link.elf.lst.txt](https://github.com/apache/mynewt-core/files/3797246/mdp-link.elf.lst.txt)

mkiiskila, 2019-11-01:
tu_printf() in your build expands to putchar(). It should've expanded to call to printf() instead. Do you know why?

brainstorm, 2019-11-01:
I was chasing exactly that right now, there's this in `tusb_common.h`:

```
#ifndef tu_printf
  #define tu_printf     printf
#endif
```

Stemming from this commit 9 days ago:

https://github.com/hathach/tinyusb/commit/661515a8072f2b8ccd601c58120ea87b96158e76#diff-1f00ae5b0cbc95f16bb2168e6ba5a32aR219

brainstorm, 2019-11-01:
But no, don't really know why is it expanding to putchar to be honest :/

Can you quickly `git clone https://github.com/chris-zen/mdp-link-mynewt && cd mdp-link-mynewt && newt build nrf52_mdp_link`?

I wonder if you can reproduce this issue...

utzig, 2019-11-01:
Not necessarily related, but I had this issue some time ago (doing some debug prints!) whenever I put `printf("");` in my code (only happened with empty string). I don't remember what `gcc` version and MCU it was, and would have to re-check if I can still happens here...

brainstorm, 2019-11-01:
Sorry to repeat myself, but wouldn't it make sense to just remove `fflush` from `stdio.h`?

mkiiskila, 2019-11-01:
Sorry, closed by accident due to mouse mishap.

mkiiskila, 2019-11-01:
I don't think removing fflush() is the answer, as that's not addressing the problem you're having. It's addressing the symptom, IMHO. I don't think the putchar() accidentally pulled in from newlib would output anything on your console.

You could add -nostdlib to your ldflags, and that would take care of removing newlib from the build. Downside is that then you will not be able to pick standalone functions from newlib which you might find useful.

adityaxavier, 2019-11-07:
Am not sure whether its relevant, however whenever I try to do a printf("\n") when compiling in Mac, it gives me this error.. 
I circumvent this problem by giving a ' ' space before the \n. Maybe that's the issue ?

brainstorm, 2019-11-09:
@adityaxavier Can you try to reproduce that with my repo/code?:

https://github.com/chris-zen/mdp-link-mynewt

I'm not sure exactly how you did that, but I would like to know more!

kasjer, 2019-12-05:
@brainstorm can you please verify that issue is fixed in latest core? #2076 should have cured your problem.

brainstorm, 2019-12-05:
Thanks @kasjer for the attention!

I just tried to recompile with the newest mynewt-core but it seems that something else is broken now:

https://github.com/apache/mynewt-core/issues/2125

So I'm not sure if this issue/part is fixed for good now :-S... yet :)

kasjer, 2020-11-24:
I think this was fixed way back by #2109

closed 2020-11-24 by kasjer

sacca97, 2023-12-05:
Problem still there on Mac M1.

Newt version:
`Apache Newt 1.12.0-dev / 24e6a29 / 2023-10-10_16:03`

Arm toolchain:
`ArmGNUToolchain/13.2.Rel1/arm-none-eabi`