apache/mynewt-core #2102: `vers: 0-latest` not accepted by newt 1.7.0

hostilefork, 2019-11-19:
Using this piece of the tutorial:

    repository.mynewt_stm32f3:
        type: github
        vers: 0-latest
        user: runtimeco
        repo: mynewt_stm32f3

There is an error message given during the `newt install -v` step:

> Error: failure parsing version for repo "mynewt_stm32f3"; Invalid version string: "master"; must be fixed (X.X.X) or floating (X[.X]-stability)

The response from `newt version` is:

    Apache Newt
        Version: 1.7.0
        Git Hash: 80bcba7
        Build Date: 2019-07-17_18:21

hostilefork, 2019-11-19:
This appears related to the issue here:

https://github.com/apache/mynewt-newt/issues/297#issuecomment-492760479

Hence the problem is that the `https://github.com/runtimeco/mynewt_stm32f3` has not had its `repository.yml` updated to follow the rule.  Unfortunately, there seems to be nothing that can be done at the repository usage site...because newt seems to be doing a kind of whole-file validation of repository.yml.

So I forked the repository to change the line, and it got past that.  The next problem is under the apache GitHub, `mcumgr`, which maps 0.0.0 to `master`

https://github.com/apache/mynewt-mcumgr/blob/8e20176e38dad0815ce6f51abad565bda747f936/repository.yml#L22

sjanc, 2019-11-19:
Hi,

hmm it looks like tutorial is very outdated as stm32f3discovery support lives now in -core.

So you don't need https://github.com/runtimeco/mynewt_stm32f3 at all. Just use "@apache-mynewt-core/hw/bsp/stm32f3discovery" as target bsp.

hostilefork, 2019-11-19:
@sjanc I'm probably going to tinker with another toolchain for the moment (as I got a different one to work).  But [the tutorial source lives in this repo](https://github.com/apache/mynewt-core/blob/master/docs/os/tutorials/STM32F303.rst) so a "This tutorial has not been updated for recent newt, please send a PR to fix it" message there seems a minimal change (e.g. would save people like me crossing it some time).

I will point out that the mcumgr was updated just 15 days ago and came up with the same complaint, so there may be more than one thing that needs fixing.

Someone might get in touch with runtimeco to add a link in the readme mentioning the deprecation and the new location.  I'm afraid I was just trying to follow what looked like a simple tutorial for a board a friend gave me with no instructions, so phrases like "target bsp" are meaningless to me.

sjanc, 2020-04-08:
Hi, tutorial is now updated

closed 2020-04-08 by sjanc