apache/mynewt-core #2125: NRF_CLOCK tinyusb function signature change

brainstorm, 2019-12-05:
Relatively recent tinyusb upstream changes seem to have broken the mynewt-core build?:

https://github.com/hathach/tinyusb/commit/4008f0d1e66dff6bb42a0fb0c682bf4bfe8e6c4a#commitcomment-36285182

brainstorm, 2019-12-05:
Perhaps related/dependent on PR https://github.com/apache/mynewt-core/pull/2107?

mlaz, 2019-12-05:
Hi Roman,

Yes, that is why, we are working to merge that PR in order to fix this
issue, since nrfx 2.0.0 breaks some drivers and features.

Miguel Azevedo

sjanc, 2020-02-28:
Hi,    nrfx was updated to 2.1.0 in https://github.com/apache/mynewt-core/pull/2206

Please test if this works for you

brainstorm, 2020-02-29:
@sjanc is the Homebrew formula updated accordingly to bump up to 2.1.0 (assuming that I installed mynewt with `brew install mynewt-newt --HEAD`?

```
$ brew reinstall mynewt-newt
==> Cloning https://github.com/apache/mynewt-newt.git
Updating /Users/romanvg/Library/Caches/Homebrew/mynewt-newt--git
==> Checking out branch master
Already on 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at 9e6fc3d Emit `size` in mfg manifest target entries
==> ./build.sh
🍺  /usr/local/Cellar/mynewt-newt/HEAD-9e6fc3d: 3 files, 9.5MB, built in 15 seconds

$ newt version
Apache Newt 1.8.0-dev / 9e6fc3d-dirty / 2020-02-29_22:02
```

utzig, 2020-02-29:
> @sjanc is the Homebrew formula updated accordingly to bump up to 2.1.0 (assuming that I installed mynewt with `brew install mynewt-newt --HEAD`?
> 
> ```
> $ brew reinstall mynewt-newt
> ==> Cloning https://github.com/apache/mynewt-newt.git
> Updating /Users/romanvg/Library/Caches/Homebrew/mynewt-newt--git
> ==> Checking out branch master
> Already on 'master'
> Your branch is up to date with 'origin/master'.
> HEAD is now at 9e6fc3d Emit `size` in mfg manifest target entries
> ==> ./build.sh
> 🍺  /usr/local/Cellar/mynewt-newt/HEAD-9e6fc3d: 3 files, 9.5MB, built in 15 seconds
> 
> $ newt version
> Apache Newt 1.8.0-dev / 9e6fc3d-dirty / 2020-02-29_22:02
> ```
> 
> I am getting the same compilation fail/results.

Upgrading `newt` only updates to tool, it does not automatically update the sources, you have to run `newt upgrade` to get the latest versions.

brainstorm, 2020-02-29:
Fair point, but still seems stuck in 1.x versions?:

```
$ newt upgrade
Downloading repository tinyusb (commit: master) from https://github.com/hathach/tinyusb.git
Skipping "mynewt_nrf52840_mdk": already upgraded (0.0.0)
Making the following changes to the project:
    upgrade apache-mynewt-core (8615b0d01d1cfbf808554fef252ae85f56098475 --> 1.7.0)
    upgrade apache-mynewt-nimble (0.0.0 --> 1.2.0)
    upgrade mcuboot (0.0.0 --> 1.3.1)
apache-mynewt-core successfully upgraded to version 1.7.0
apache-mynewt-nimble successfully upgraded to version 1.2.0
mcuboot successfully upgraded to version 1.3.1
tinyusb successfully upgraded to version 0.0.0
```

utzig, 2020-03-01:
> Fair point, but still seems stuck in 1.x versions?:
>
> ```
> $ newt upgrade
> Downloading repository tinyusb (commit: master) from https://github.com/hathach/tinyusb.git
> Skipping "mynewt_nrf52840_mdk": [...]

You have to update your `project.yml` to use `vers: 0-dev` instead of `vers: 1-latest`

brainstorm, 2020-03-01:
Thanks, makes sense, here's the result:

```
$ newt upgrade
Skipping "apache-mynewt-mcumgr": already upgraded (0.0.0)
Skipping "mynewt_nrf52840_mdk": already upgraded (0.0.0)
Skipping "tinyusb": already upgraded (0.0.0)
Making the following changes to the project:
    upgrade apache-mynewt-core (1.7.0 --> 0.0.0)
    upgrade apache-mynewt-nimble (1.2.0 --> 0.0.0)
    upgrade mcuboot (1.3.1 --> 0.0.0)
apache-mynewt-core successfully upgraded to version 0.0.0
apache-mynewt-nimble successfully upgraded to version 0.0.0
mcuboot successfully upgraded to version 0.0.0

$  newt build nrf52_mdp_link
WARNING: apache-mynewt-core: Repo version missing from compatibility map
Building target targets/nrf52_mdp_link
Compiling repos/mynewt_nrf52840_mdk/hw/bsp/nrf52840_mdk/src/sbrk.c
Compiling repos/mynewt_nrf52840_mdk/hw/bsp/nrf52840_mdk/src/hal_bsp.c
Compiling apps/mdp-link/src/main.c
Compiling apps/mdp-link/src/usb_descriptors.c
Assembling repos/mynewt_nrf52840_mdk/hw/bsp/nrf52840_mdk/src/arch/cortex_m4/gcc_startup_nrf52840.s
Assembling repos/mynewt_nrf52840_mdk/hw/bsp/nrf52840_mdk/src/arch/cortex_m4/gcc_startup_nrf52_split.s
Error: apps/mdp-link/src/main.c: In function 'usb_hardware_init':
apps/mdp-link/src/main.c:230:9: error: passing argument 1 of 'nrf_power_int_enable' makes pointer from integer without a cast [-Werror=int-conversion]
         NRF_POWER_INT_USBDETECTED_MASK |
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
         NRF_POWER_INT_USBREMOVED_MASK  |
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
         NRF_POWER_INT_USBPWRRDY_MASK);
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from apps/mdp-link/src/main.c:36:
repos/apache-mynewt-core/hw/mcu/nordic/src/ext/nrfx/hal/nrf_power.h:899:62: note: expected 'NRF_POWER_Type *' {aka 'struct <anonymous> *'} but argument is of type 'int'
 NRF_STATIC_INLINE void nrf_power_int_enable(NRF_POWER_Type * p_reg, uint32_t mask)
                                             ~~~~~~~~~~~~~~~~~^~~~~
apps/mdp-link/src/main.c:229:3: error: too few arguments to function 'nrf_power_int_enable'
   nrf_power_int_enable(
   ^~~~~~~~~~~~~~~~~~~~
In file included from apps/mdp-link/src/main.c:36:
repos/apache-mynewt-core/hw/mcu/nordic/src/ext/nrfx/hal/nrf_power.h:899:24: note: declared here
 NRF_STATIC_INLINE void nrf_power_int_enable(NRF_POWER_Type * p_reg, uint32_t mask)
                        ^~~~~~~~~~~~~~~~~~~~
apps/mdp-link/src/main.c: In function 'POWER_CLOCK_IRQHandler':
apps/mdp-link/src/main.c:254:22: error: too few arguments to function 'nrf_power_int_enable_get'
   uint32_t enabled = nrf_power_int_enable_get();
                      ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from apps/mdp-link/src/main.c:36:
repos/apache-mynewt-core/hw/mcu/nordic/src/ext/nrfx/hal/nrf_power.h:909:28: note: declared here
 NRF_STATIC_INLINE uint32_t nrf_power_int_enable_get(NRF_POWER_Type const * p_reg)
                            ^~~~~~~~~~~~~~~~~~~~~~~~
apps/mdp-link/src/main.c:257:37: error: passing argument 1 of 'nrf_power_event_get_and_clear' makes pointer from integer without a cast [-Werror=int-conversion]
       nrf_power_event_get_and_clear(NRF_POWER_EVENT_USBDETECTED))
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from apps/mdp-link/src/main.c:36:
repos/apache-mynewt-core/hw/mcu/nordic/src/ext/nrfx/hal/nrf_power.h:882:72: note: expected 'NRF_POWER_Type *' {aka 'struct <anonymous> *'} but argument is of type 'int'
 NRF_STATIC_INLINE bool nrf_power_event_get_and_clear(NRF_POWER_Type *  p_reg,
                                                      ~~~~~~~~~~~~~~~~~~^~~~~
apps/mdp-link/src/main.c:257:7: error: too few arguments to function 'nrf_power_event_get_and_clear'
       nrf_power_event_get_and_clear(NRF_POWER_EVENT_USBDETECTED))
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from apps/mdp-link/src/main.c:36:
repos/apache-mynewt-core/hw/mcu/nordic/src/ext/nrfx/hal/nrf_power.h:882:24: note: declared here
 NRF_STATIC_INLINE bool nrf_power_event_get_and_clear(NRF_POWER_Type *  p_reg,
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
apps/mdp-link/src/main.c:263:37: error: passing argument 1 of 'nrf_power_event_get_and_clear' makes pointer from integer without a cast [-Werror=int-conversion]
       nrf_power_event_get_and_clear(NRF_POWER_EVENT_USBREMOVED))
                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from apps/mdp-link/src/main.c:36:
repos/apache-mynewt-core/hw/mcu/nordic/src/ext/nrfx/hal/nrf_power.h:882:72: note: expected 'NRF_POWER_Type *' {aka 'struct <anonymous> *'} but argument is of type 'int'
 NRF_STATIC_INLINE bool nrf_power_event_get_and_clear(NRF_POWER_Type *  p_reg,
                                                      ~~~~~~~~~~~~~~~~~~^~~~~
apps/mdp-link/src/main.c:263:7: error: too few arguments to function 'nrf_power_event_get_and_clear'
       nrf_power_event_get_and_clear(NRF_POWER_EVENT_USBREMOVED))
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from apps/mdp-link/src/main.c:36:
repos/apache-mynewt-core/hw/mcu/nordic/src/ext/nrfx/hal/nrf_power.h:882:24: note: declared here
 NRF_STATIC_INLINE bool nrf_power_event_get_and_clear(NRF_POWER_Type *  p_reg,
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
apps/mdp-link/src/main.c:269:37: error: passing argument 1 of 'nrf_power_event_get_and_clear' makes pointer from integer without a cast [-Werror=int-conversion]
       nrf_power_event_get_and_clear(NRF_POWER_EVENT_USBPWRRDY))
[... cut ...]
apps/mdp-link/src/main.c:269:7: error: too few arguments to function 'nrf_power_event_get_and_clear'
       nrf_power_event_get_and_clear(NRF_POWER_EVENT_USBPWRRDY))
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from apps/mdp-link/src/main.c:36:
repos/apache-mynewt-core/hw/mcu/nordic/src/ext/nrfx/hal/nrf_power.h:882:24: note: declared here
 NRF_STATIC_INLINE bool nrf_power_event_get_and_clear(NRF_POWER_Type *  p_reg,
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
```