apache/mynewt-core #2185: question of mynewt setup

li-yizhi, 2020-02-08:
I'm a beginner of mynewt. When I set up mynewt to this step [Use a newt wrapper script](http://mynewt.apache.org/latest/get_started/docker.html#use-a-newt-wrapper-script)
I confused:
1.

> Create the following file, name it newt
The file's name is newt, what's it suffix name?
2.

> make it executable
How to do this step?
3.

> put it in your path
What is /your path/ ? Docker's Installation path or any other path.

utzig, 2020-02-08:
Those are all OS specific concepts that are not Mynewt related, which OS are you using?

li-yizhi, 2020-02-09:
> Those are all OS specific concepts that are not Mynewt related, which OS are you using?

win 10

cqexbesd, 2020-06-03:
The docs are written assuming a UNIX like environment. If you have the Linux subsystem for Windows 10 you could probably use them unmodified however I don’t have any experience with Windows 10 so I’m just guessing.

Assuming that the Linux subsystem will make your W10 machine behave like UNIX then:

1. No suffix is required
2. chmod a+x file
3. The path is the list of directories the shell looks for commands. How to set it depends on your shell but assuming you are using `bash` then `export PATH=${PATH}:/path/to/new/directory`

I would recommend having a read of some documentation about basic UNIX commands and the Linux subsystem though, as @utzig said, these aren’t mynewt concepts.

li-yizhi, 2020-07-02:
> The docs are written assuming a UNIX like environment. If you have the Linux subsystem for Windows 10 you could probably use them unmodified however I don’t have any experience with Windows 10 so [...]

Thanks a lot! It's very helpful!

closed 2022-04-13 by sjanc