apache/mynewt-core #2189: Could not resolve app package: @mcuboot/boot/mynewt

Thilanka97, 2020-02-12:
hey I am facing this problem with blinky project.
 _**Could not resolve app package: @mcuboot/boot/mynewt**_

 I tried _**newt upgrade -f**_  but I am getting following error with that cmd. 
_**Error: usage: git rev-list [OPTION] <commit-id>... [ -- paths... ]**_

Also when I create the targets, inside the repo folder of the project the folders _"mcuboot" ,_ _"apache-mynewt-mcumgr"_ and _"apache-mynewt-nimble"_ folders are not created.

What is wrong here. I have already installed git as well.  Please help me to solve this problem. 

Thanks in advance!

sjanc, 2020-02-12:
Hi,

which newt version and what OS do you use?

utzig, 2020-02-12:
Check `git --version` and do `newt -ldebug upgrade -v` to get some debugging, but you probably need to upgrade `newt` due to some issue with submodules in mcuboot.

w277277w, 2020-03-22:
HI，I had the same problem.
How did you solve the problem?

sjanc, 2020-04-17:
is this stil an issue with 1.8 ?

w277277w, 2020-04-18:
it is no issue with 1.8, It's wrong that the project upgrade without '--escape=false' on windows, thank you

sjanc, 2020-04-20:
great!

closed 2020-04-20 by sjanc