apache/mynewt-core #2246: newt upgrade failed - Error: usage: git rev-list [OPTION] <commit-id>... [ -- paths... ]

w277277w, 2020-03-22:
While creating this mynewt project:  [Blinky on nRF52 DK](http://mynewt.apache.org/latest/tutorials/blinky/nRF52.html) on windown10

The version of mynewt:
```
Surface@DESKTOP-D0I1LH5 MINGW64 ~/dev/myproj
$ newt version
Apache Newt
   Version: 1.6.0
  Git Hash: da3758b
Build Date: 2019-04-02_22:49
```
I followed the document [Blinky on nRF52 DK](http://mynewt.apache.org/latest/tutorials/blinky/nRF52.html) step by step 
The error occurred when the mynewt-core was download completed
```
Surface@DESKTOP-D0I1LH5 MINGW64 ~/dev/myproj
$ newt upgrade
Downloading repository mynewt-core (commit: master) from https://github.com/apache/mynewt-core.git
....
Error: usage: git rev-list [OPTION] <commit-id>... [ -- paths... ]
  limiting output:
    --max-count=<n>
    --max-age=<epoch>
    --min-age=<epoch>
    --sparse
...
```
Please help me, thanks very much!
Detial of the error
```
Surface@DESKTOP-D0I1LH5 MINGW64 ~/dev/myproj
$ newt upgrade -ldebug
2020/03/22 17:50:50.452 [DEBUG] Searching for project file D:/MyDevelopment/msys64/home/Surface/dev/myproj/project.yml
2020/03/22 17:50:50.572 [DEBUG] Read config file: D:\MyDevelopment\msys64\home\Surface\dev\myproj\project.yml
...
...
...
2020/03/22 17:51:07.280 [DEBUG] Read config file: D:\MyDevelopment\msys64\home\Surface\dev\myproj\repos\apache-mynewt-core\util\taskpool\syscfg.yml
2020/03/22 17:51:07.396 [DEBUG] D:/MyDevelopment/Git/cmd/git.exe remote get-url origin
2020/03/22 17:51:07.532 [DEBUG] o=https://github.com/apache/mynewt-core.git

2020/03/22 17:51:07.629 [DEBUG] D:/MyDevelopment/Git/cmd/git.exe fetch --tags
2020/03/22 17:51:22.333 [DEBUG] o=
2020/03/22 17:51:22.455 [DEBUG] D:/MyDevelopment/Git/cmd/git.exe merge-base master master
2020/03/22 17:51:22.575 [DEBUG] o=ebae463b8c52ba880be840f1961b259b433d152d

2020/03/22 17:51:22.691 [DEBUG] D:/MyDevelopment/Git/cmd/git.exe show-ref --verify --quiet refs/heads/master
2020/03/22 17:51:22.792 [DEBUG] o=
2020/03/22 17:51:22.908 [DEBUG] D:/MyDevelopment/Git/cmd/git.exe rev-parse --abbrev-ref --symbolic-full-name master@\{u\}
2020/03/22 17:51:23.022 [DEBUG] o=master@\{u\}

2020/03/22 17:51:23.022 [DEBUG] Fetching file repository.yml to D:/MyDevelopment/msys64/home/Surface/dev/myproj/repos/.configs/apache-mynewt-core/repository.yml
2020/03/22 17:51:23.134 [DEBUG] D:/MyDevelopment/Git/cmd/git.exe show master@\\{u\\}:repository.yml
2020/03/22 17:51:23.261 [DEBUG] o=
2020/03/22 17:51:23.262 [DEBUG] Read config file: D:\MyDevelopment\msys64\home\Surface\dev\myproj\repos\.configs\apache-mynewt-core\repository.yml
2020/03/22 17:51:23.263 [DEBUG] Read config file: D:\MyDevelopment\msys64\home\Surface\dev\myproj\repos\apache-mynewt-core\version.yml
2020/03/22 17:51:23.380 [DEBUG] D:/MyDevelopment/Git/cmd/git.exe rev-parse HEAD
2020/03/22 17:51:23.493 [DEBUG] o=ebae463b8c52ba880be840f1961b259b433d152d

2020/03/22 17:51:24.150 [DEBUG] D:/MyDevelopment/Git/cmd/git.exe merge-base ebae463b8c52ba880be840f1961b259b433d152d ebae463b8c52ba880be840f1961b259b433d152d
2020/03/22 17:51:24.309 [DEBUG] o=ebae463b8c52ba880be840f1961b259b433d152d

2020/03/22 17:51:24.429 [DEBUG] D:/MyDevelopment/Git/cmd/git.exe show-ref --verify --quiet refs/heads/ebae463b8c52ba880be840f1961b259b433d152d
2020/03/22 17:51:24.539 [DEBUG] o=
2020/03/22 17:51:24.539 [DEBUG] err=exit status 1
2020/03/22 17:51:24.655 [DEBUG] D:/MyDevelopment/Git/cmd/git.exe rev-parse ebae463b8c52ba880be840f1961b259b433d152d
2020/03/22 17:51:24.776 [DEBUG] o=ebae463b8c52ba880be840f1961b259b433d152d

2020/03/22 17:51:24.884 [DEBUG] D:/MyDevelopment/Git/cmd/git.exe for-each-ref --format=%(refname:short) --points-at ebae463b8c52ba880be840f1961b259b433d152d
2020/03/22 17:51:25.020 [DEBUG] o=master
origin/HEAD
[... 77 lines cut ...]
        C:/msys64/tmp/mynewt.BKGAmIvGWn/src/mynewt.apache.org/newt/newt/install/install.go:712 +0x82
mynewt.apache.org/newt/newt/install.(*Installer).Upgrade(0xc0001a3e20, 0xc0001a60a0, 0x1, 0x1, 0xc000070000, 0x0, 0x0)
        C:/msys64/tmp/mynewt.BKGAmIvGWn/src/mynewt.apache.org/newt/newt/install/install.go:836 +0x90
mynewt.apache.org/newt/newt/project.(*Project).InstallIf(0xc00010a400, 0x1, 0xc00049baa0, 0xc00049baa0, 0x0)
        C:/msys64/tmp/mynewt.BKGAmIvGWn/src/mynewt.apache.org/newt/newt/project/project.go:279 +0x135
mynewt.apache.org/newt/newt/cli.upgradeRunCmd(0xc00010d900, 0xc00003cee0, 0x0, 0x1)
        C:/msys64/tmp/mynewt.BKGAmIvGWn/src/mynewt.apache.org/newt/newt/cli/project_cmds.go:122 +0xad
mynewt.apache.org/newt/vendor/github.com/spf13/cobra.(*Command).execute(0xc00010d900, 0xc00003cea0, 0x1, 0x1, 0xc00010d900, 0xc00003cea0)
        C:/msys64/tmp/mynewt.BKGAmIvGWn/src/mynewt.apache.org/newt/vendor/github.com/spf13/cobra/command.go:766 +0x2d3
mynewt.apache.org/newt/vendor/github.com/spf13/cobra.(*Command).ExecuteC(0xc000070c80, 0x7, 0x1f, 0x7f43df)
        C:/msys64/tmp/mynewt.BKGAmIvGWn/src/mynewt.apache.org/newt/vendor/github.com/spf13/cobra/command.go:852 +0x304
mynewt.apache.org/newt/vendor/github.com/spf13/cobra.(*Command).Execute(0xc000070c80, 0x7f43df, 0x7)
        C:/msys64/tmp/mynewt.BKGAmIvGWn/src/mynewt.apache.org/newt/vendor/github.com/spf13/cobra/command.go:800 +0x32
main.main()
        C:/msys64/tmp/mynewt.BKGAmIvGWn/src/mynewt.apache.org/newt/newt/newt.go:184 +0x178

goroutine 5 [syscall]:
os/signal.signal_recv(0x0)
        C:/msys64/mingw64/lib/go/src/runtime/sigqueue.go:139 +0xa3
os/signal.loop()
        C:/msys64/mingw64/lib/go/src/os/signal/signal_unix.go:23 +0x29
created by os/signal.init.0
        C:/msys64/mingw64/lib/go/src/os/signal/signal_unix.go:29 +0x48

```

w277277w, 2020-03-23:
@utzig Execute me, could you help me?

sjanc, 2020-03-23:
Hi, this is probably due to broken escaping of "@" on windows. Please try with 1.7 (or 1.8-rc1) with newt upgrade --escape=true (or false)

w277277w, 2020-03-26:
@sjanc Thank you for your mind. The issue was sloved

sjanc, 2020-03-26:
Great!

closed 2020-03-26 by sjanc