apache/mynewt-core #2273: RFC - ADC API not abstract enough for a test app.

mlaz, 2020-04-17:
Currently the ADC API lacks on device and channel configuration data structures. This led us to having drivers which work in different fashions and use different configuration data structures. It made us deal with manufacturer SDK data structures on app code and stopped us from having an adc_test app (like we do with pwm_test) which besides being example code is helpful to test the drivers during manufacturer SDK code updates.

This issue aims to find an ADC API witch abstracts as much as possible the hardware thus allowing more homogenous usage across drivers. 

At this point I did a few [changes](https://github.com/mlaz/mynewt-core/tree/adc_api_changes) on the ADC API  and nRF52 driver so it works with the test app.
The test app is based on the one we have on this [tutorial](https://mynewt.apache.org/latest/tutorials/sensors/nrf52_adc.html) and it can improved.

Please review and comment.

caspermeijn, 2020-04-21:
I think this is an interesting idea. I already found it strange that the nrf52 ADC drivers where to MCU specific. 
- I think it would be useful to see more then one driver use this api, then you will see if this model matches a different MCU. 
- I think it is an open question what to do if a feature is not supported by the hardware. For example an ACQ_TIME of 50 us will now result in 10 us selected, but 40us would be more appropriate.
- I added some inline comments to the commit, however these where not linked to this issue as this is not a PR.

shanyechungu, 2020-07-23:
Please also include example code of multiple channel scan mode for nrf52 SAADC. Thanks,