apache/mynewt-core #2354: LoRa with RFM95

danielkucera, 2020-08-16:
Hello,

I'm unable to join with RFM95 radio module. I defined my own BSP, tried both sx1276 and sx1272 drivers but without luck. The result is always the following:
```
400851 Attempting to join...
compat> Join cb. status=7 attempts=1
```
The same hardware works with mbed framework. 
The signal seems to be very week in comparison with mbed, See attached gqrx screenshots:
mynewt sx1272:
[image]

mynewt sx1276:
[image]

mbed:
[image]

danielkucera, 2021-02-08:
I managed to fix the signal strength issue with:
```
SX1276_HF_USE_PA_BOOST: 1
```
[image]

But it is still not registering using `las_join 1`. I don't see any RX packets in my gateway. I'll keep investigating.

danielkucera, 2021-02-08:
Ok, the last missing value to correctly connect to TTN was:
```
    LORA_NODE_PUBLIC_NWK: 1
```
so the final config is:
```
    SPI_0_MASTER: 1
    SPI_0_MASTER_PIN_SCK: 30
    SPI_0_MASTER_PIN_MOSI: 2
    SPI_0_MASTER_PIN_MISO: 3

    # repos/apache-mynewt-core/net/lora/node/include/node/lora_band.h
    LORA_NODE: 1
    LORA_NODE_REGION: 6
    LORA_MAC_TIMER_NUM: 0
    LORA_NODE_PUBLIC_NWK: 1

    # lora pins
    #SX1276
    SX1276_SPI_IDX: 0
    SX1276_SPI_CS_PIN: 29 # NSS?
    SX1276_HF_USE_PA_BOOST: 1
```
and pins in `bsp.h`:
```
#define SX1276_DIO0         (31)
#define SX1276_DIO1         (4)
#define SX1276_DIO2         (5)
#define SX1276_DIO3         (26)
#define SX1276_DIO4         (28)
#define SX1276_DIO5         (25)
#define SX1276_NRESET       (27)
```
But I have to say that the documentation is very poor. :-1:

tcpipchip, 2022-11-11:
i am having the same ISSUE :(
[image]

danielkucera, 2022-11-14:
As I wrote, it is.

closed 2022-11-14 by danielkucera

tcpipchip, 2022-11-14:
i am getting a progresss here...
i was using SPI0...and software was on SPI1

tcpipchip, 2022-11-14:
are you using wich processor? And do you created the bootloader and app ?