apache/mynewt-core #2374: Error: bsp_download.sh: No such file or directory

krzysztofdabek-semcon, 2020-09-10:
I am following the Hello tutorial for MyNewt and while executing command:
`newt load nrf 52_boot`
I am getting an error:
`Error: /bin/bash: C:\Workspace\myproj\repos\apache-mynewt-core\hw\bsp\nordic_pca10040\nordic_pca10040_download.sh: No such file or directory`
even though the file exists.

I am working on Windows 10 and that happens in CMD, GitBash, MSYS bash and VSCode bash.

nkaje, 2020-09-10:
Not sure if there is any path related issue on windows. It may help if you paste the output with debug enabled
`newt load -ldebug nrf52...`

krzysztofdabek-semcon, 2020-09-11:
> Not sure if there is any path related issue on windows. It may help if you paste the output with debug enabled
> `newt load -ldebug nrf52...`

Here is the output with debug enabled:
https://semcon-my.sharepoint.com/:t:/p/krzysztof_dabek/EUbQbm37t7NHryZgVF-q8tcBdc8xFDwfk9ZHLpJLDUkLEA?e=b7jrk3

nkaje, 2020-09-11:
From your logs, it seems like you are missing
https://github.com/apache/mynewt-newt/pull/403. That change is for executing .sh scripts with bash on windows.

You could try building the latest mynewt-newt (See: https://github.com/apache/mynewt-newt/blob/master/INSTALLING.md)

krzysztofdabek-semcon, 2020-09-14:
Same issue on the latest version 1.9.0 dev

nkaje, 2020-09-14:
```
commit 8330e7f8bd6570e2bccb73377b115ab78541933d
Author: J. Ipanienko <j.ipanienko@mexicomail.com>
Date:   Wed Feb 19 13:19:16 2020 -0800

    Execute .sh scripts with bash on windows
```
This commit is newer than 1.9.0-dev (c5d1e422bfa335b3b58589eca2401ec3616e889e) release. I suspect you may have to try with the HEAD of master branch. @jipanienko could you please see if this issue is addressed by your fix?

krzysztofdabek-semcon, 2020-09-14:
Still no luck. Anything I could have missed?

sjanc, 2020-09-18:
I've just tested with msys (64 bit) and it seems to be working just fine on Windows 10.
It is weird that path in error message from bash is with windows slash instead of unix backslash...

- How do you build newt from master ? I'm using go installed in msys via pacman (not Windows one)
- run msys64 shell with admin privileges (otherwise symlinks are not working correctly)

jipanienko, 2020-09-21:
@nkaje no, unrelated problem.
More like strange (non msys) bash in PATH

closed 2022-04-13 by sjanc