apache/mynewt-core #2388: newt install error - no remote upstream for branch master

pchat-imm, 2020-10-07:
I tried to use command `newt install` and `newt upgrade`. However, both of them give error 

> Error: Error downloading repository apache-mynewt-core: No remote upstream for branch "master"

This is the capture of my screen

> $ newt install -v
[apache-mynewt-core]:
Downloading repository description
Downloading repository mynewt-core (commit: master) from https://github.com/apache/mynewt-core.git
Cloning into 'C:\msys64\usr\bin\tmp\newt-repo758994003'...
remote: Enumerating objects: 11, done.
remote: Counting objects: 100% (11/11), done.
remote: Compressing objects: 100% (10/10), done.
remote: Total 116628 (delta 1), reused 1 (delta 0), pack-reused 116617
Receiving objects: 100% (116628/116628), 167.07 MiB | 1.11 MiB/s, done.
Resolving deltas: 100% (71692/71692), done.
Updating files: 100% (6497/6497), done.
Error: Error downloading repository apache-mynewt-core: No remote upstream for branch "master"

> $ newt upgrade
Downloading repository mynewt-core (commit: master) from https://github.com/apache/mynewt-core.git
Error: Error downloading repository apache-mynewt-core: No remote upstream for branch "master"

please help me solve this problem

sjanc, 2020-10-07:
What newt version do you use?

pchat-imm, 2020-10-07:
@sjanc I tried newt version 1.6.0.
Also, i tried both --escape=false and --escape=true
If you have more suggestion please help me
Thank you

sjanc, 2020-10-08:
I'd try with newt 1.8 as there were some further fixes for windows support. Also what shell do you use under windows? I'm testing on bash from msys  and it seems to work OK in Windows 10.

pchat-imm, 2020-10-09:
Oh that works like a charm! Thank you very much.
I changed to newt 1.8 as you suggested. And work on bash from msys.

Thank you very much

sjanc, 2020-10-09:
Great!

closed 2020-10-09 by sjanc