apache/mynewt-core #2473: Priority violations

danielkucera, 2021-02-06:
Latest master fails to compile with throwing following error:
```
Building target targets/nrf52_boot
Error: Priority violations detected (Packages can only override settings defined by packages of lower priority):
    Package: @apache-mynewt-core/hw/mcu/nordic/nrf52xxx overriding setting: OS_TICKS_PER_SEC defined by @apache-mynewt-core/kernel/os

Setting history (newest -> oldest):
    OS_TICKS_PER_SEC: [@apache-mynewt-core/hw/mcu/nordic/nrf52xxx:128, @apache-mynewt-core/kernel/os:]
```
How do I fix this?

andrzej-kaczmarek, 2021-02-06:
You need to use latest newt tool (see https://github.com/apache/mynewt-newt/pull/423).

navan93, 2021-02-09:
By latest do you mean version 1.8.0? I alread have Apache Newt 1.8.0 / unknown / 2020-06-23_13:52. Still getting the same error as above.

danielkucera, 2021-02-09:
1.8.0 seems to be not enough. A new release would be helpful.

mlaz, 2021-02-09:
@danielkucera you need to be using master. BTW, @andrzej-kaczmarek maybe we could start tagging the PRs that break build with non-latest master and cross referencing PRs between core and newt, what do you think?

sjanc, 2021-02-15:
we could, although general rule of thumb would be use tools from master when playing with core master branch (same goes for nimble repo)

sjanc, 2021-03-31:
@danielkucera you may wanna check with 1.9-rc1  https://dist.apache.org/repos/dist/dev/mynewt/apache-mynewt-1.9.0/rc1/

sjanc, 2021-04-19:
is this stil an issue with 1.9 ?

danielkucera, 2021-04-26:
It seems yes but now I'm stuck on another error #2583

sjanc, 2024-09-06:
reopen if needed

closed 2024-09-06 by sjanc