apache/mynewt-core #2485: STM32F4-Discovery Blinky build issue - ".text will not fit in region FLASH"

stuffmatic, 2021-02-17:
I ran into a build issue when doing [the Blinky tutorial](https://mynewt.apache.org/latest/tutorials/blinky/blinky_stm32f4disc.html) with my [STM32F4-Discovery board](https://www.st.com/en/evaluation-tools/stm32f4discovery.html). `newt build stm32f4disc_boot` fails with the following error

```
Error: /usr/local/Caskroom/gcc-arm-embedded/10-2020-q4-major/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: [redacted]/mynewt-sandbox/bin/targets/stm32f4disc_boot/app/@mcuboot/boot/mynewt/mynewt.elf section `.text' will not fit in region `FLASH'
/usr/local/Caskroom/gcc-arm-embedded/10-2020-q4-major/gcc-arm-none-eabi-10-2020-q4-major/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: region `FLASH' overflowed by 348 bytes
collect2: error: ld returned 1 exit status
```

The error message above is when using the master branch. 1.8.0 gives me the same error, but with a different (smaller) number of bytes in `region 'FLASH' overflowed by xxx bytes`.

Naively changing `FLASH (rx) :  ORIGIN = 0x08000000, LENGTH = 16K` to `FLASH (rx) :  ORIGIN = 0x08000000, LENGTH = 17K` in [boot-stm32f4discovery.ld](https://github.com/apache/mynewt-core/blob/master/hw/bsp/stm32f4discovery/boot-stm32f4discovery.ld) made app build and run on my board, but I suspect this does not count as a solution.

I'm running macOS Big Sur 11.2.

mlaz, 2021-02-17:
Hi @stuffmatic try disabling some features on mcuboot. Look at what's enabled by default on mcuboot's syscfg files and disable it on your target's syscfg.

stuffmatic, 2021-02-17:
Thanks for the tip @mlaz ! I'll have a look.

stuffmatic, 2021-03-01:
I couldn't find any obvious things to disable in my target's syscfg... Any pointers would be much appreciated.

utzig, 2021-03-01:
> I couldn't find any obvious things to disable in my target's syscfg... Any pointers would be much appreciated.

Try adding this before the `build` command:
```
newt target set stm32f4disc_boot syscfg=BOOTUTIL_USE_MBED_TLS=0:BOOTUTIL_USE_TINYCRYPT=1:TIMER_0=0:OS_CPUTIME_TIMER_NUM="-1":LOG_CONSOLE=0
```

stuffmatic, 2021-03-02:
For posterity: the size decrease using `BOOTUTIL_...` was insufficient, but `TIMER_0=0` or `OS_CPUTIME_TIMER_NUM=-1` worked. Thanks!

closed 2021-03-02 by stuffmatic