apache/mynewt-core #2507: Error creating initial 'myproj' project

jcwren, 2021-02-26:
Following the instructions for creating a first project, and get this error when running `newt upgrade`

Note that I did not change the `vers` field in the `project.yml` file. The instructions [here](http://mynewt.apache.org/latest/get_started/project_create.html#fetching-the-mynewt-source-repository-and-dependencies) don't match up with what's in the file.

```
jcw@debresc:~/Projects/myproj$ newt version
Apache Newt
   Version: 1.7.0
  Git Hash: unknown
Build Date: 2019-09-03_13:44

jcw@debresc:~/Projects$ newt new myproj
Downloading project skeleton from apache/mynewt-blinky...
Downloading repository mynewt-blinky (commit: mynewt_1_7_0_tag) from https://github.com/apache/mynewt-blinky.git
Installing skeleton in myproj...
Project myproj successfully created.
```
```
jcw@debresc:~/Projects$ cd myproj/
jcw@debresc:~/Projects/myproj$ cat project.yml
#
# [snip comments for brevity]
#

project.name: "my_project"

project.repositories:
    - apache-mynewt-core

# Use github's distribution mechanism for core ASF libraries.
# This provides mirroring automatically for us.
#
repository.apache-mynewt-core:
    type: github
    vers: 1.7.0
    user: apache
    repo: mynewt-core
```
```
jcw@debresc:~/Projects/myproj$ newt upgrade
Downloading repository mynewt-core (commit: master) from https://github.com/apache/mynewt-core.git
Downloading repository mynewt-nimble (commit: master) from https://github.com/apache/mynewt-nimble.git
Downloading repository mcuboot (commit: master) from https://github.com/JuulLabs-OSS/mcuboot.git
Downloading repository mynewt-mcumgr (commit: master) from https://github.com/apache/mynewt-mcumgr.git
Making the following changes to the project:
    install apache-mynewt-core (1.7.0)
    install apache-mynewt-nimble (1.2.0)
    install mcuboot (1.3.1)
apache-mynewt-core successfully upgraded to version 1.7.0
apache-mynewt-nimble successfully upgraded to version 1.2.0
Error: Error updating "mcuboot": error: The following untracked working tree files would be overwritten by checkout:
        ext/mbedtls/include/mbedtls/check_config.h
        ext/mbedtls/include/mbedtls/config.h
Please move or remove them before you switch branches.
Aborting

jcw@debresc:~/Projects/myproj$
```

jcwren, 2021-02-26:
As a test, I created a new project, changed the `vers: 1.7.0` to `vers: 1-latest` and got this error:
```
jcw@debresc:~/Projects/myproj$ newt upgrade
Error: Repo "apache-mynewt-core" contains invalid version requirement: 1-lastest (Unknown stability (lastest) in version 1-lastest)
jcw@debresc:~/Projects/myproj$ vi project.yml
jcw@debresc:~/Projects/myproj$ newt upgrade
Downloading repository mynewt-core (commit: master) from https://github.com/apache/mynewt-core.git
Downloading repository mynewt-nimble (commit: master) from https://github.com/apache/mynewt-nimble.git
Downloading repository mcuboot (commit: master) from https://github.com/JuulLabs-OSS/mcuboot.git
Downloading repository mynewt-mcumgr (commit: master) from https://github.com/apache/mynewt-mcumgr.git
Making the following changes to the project:
    install apache-mynewt-core (1.8.0)
    install apache-mynewt-mcumgr (0.1.0)
    install apache-mynewt-nimble (1.3.0)
    install mcuboot (1.5.0)
Error: This version of newt (1.7.0) is incompatible with your version of the apache-mynewt-core repo (1.8.0); Please upgrade your newt tool to version 1.8.0
This version of newt (1.7.0) is incompatible with your version of the apache-mynewt-nimble repo (1.3.0); Please upgrade your newt tool to version 1.8.0
jcw@debresc:~/Projects/myproj$
```
I had installed newt as per the instructions.
```
 2066  wget -qO - https://raw.githubusercontent.com/JuulLabs-OSS/debian-mynewt/master/mynewt.gpg.key | sudo apt-key add -
 $ sudo tee /etc/apt/sources.list.d/mynewt.list <<EOF
deb https://raw.githubusercontent.com/JuulLabs-OSS/debian-mynewt/master latest main
EOF
 $ sudo apt-get update -q -q
 $ sudo apt-get install newt
 $ sudo apt-get install gcc-multilib

utzig, 2021-02-26:
Unfortunately this is a known issue with newt 1.7.0, you should use at least 1.8.0. It seems that the Ubuntu package was not updated but you can always get the latest release with:

```
go get mynewt.apache.org/newt/newt
```
It should work as long as you have a fairly recent `go` tooling and export `$GOPATH/bin` on `$PATH`.

jcwren, 2021-02-26:
I'll give that a try. I know next to nothing about `go`. Do I need a `sudo` with that, or where is it going to install `newt`?

utzig, 2021-02-26:
> I'll give that a try. I know next to nothing about `go`. Do I need a `sudo` with that, or where is it going to install `newt`?

```
export GOPATH=$HOME/go
export PATH=$PATH:$GOPATH/bin
go get mynewt.apache.org/newt/newt
```

no `sudo` required.

utzig, 2021-02-26:
@jcwren I updated the debian repos, now you can apt-get update and upgrade and it should update to newt 1.8.0. If it works and `newt update` works (it should!), please close this issue.

jcwren, 2021-02-26:
Excellent. I'm able to create a new project, update it, build it, and run it.  

There's still one issue, I can open this as a new one, if you prefer.  The `newt install` command returns the following:
```
Error: unknown command "install" for "newt"
Run 'newt --help' for usage.
jcw@debresc:~/Projects/myproj$
```

The below is from the README.md file that's created.

---------- 

1. Download and install Apache Newt.

You will need to download the Apache Newt tool, as documented in the [Getting Started Guide](https://mynewt.apache.org/latest/get_started/index.html).

2. Download the Apache Mynewt Core package (executed from the blinky directory).

```no-highlight
    $ newt install
```

3. Build the blinky app for the sim platform using the "my_blinky_sim" target
(executed from the blinky directory).

```no-highlight
    $ newt build my_blinky_sim
```

utzig, 2021-02-26:
> Excellent. I'm able to create a new project, update it, build it, and run it.
> 
> There's still one issue, I can open this as a new one, if you prefer. The `newt install` command returns the following:
> 
> ```
> Error: unknown command "install" for "newt"
> Run 'newt --help' for usage.
> jcw@debresc:~/Projects/myproj$
> ```
> 
> The below is from the README.md file that's created.
> 
> 1. Download and install Apache Newt.
> 
> You will need to download the Apache Newt tool, as documented in the [Getting Started Guide](https://mynewt.apache.org/latest/get_started/index.html).
> 
> 1. Download the Apache Mynewt Core package (executed from the blinky directory).
> 
> ```
>     $ newt install
> ```
> 
> 1. Build the blinky app for the sim platform using the "my_blinky_sim" target
>    (executed from the blinky directory).
> 
> ```
>     $ newt build my_blinky_sim
> ```

Ah, `install` existed before, but was removed and `upgrade` can do everything now, so the documentation got behind, you can PR a fix here if you want: https://github.com/apache/mynewt-blinky/blob/master/README.md. Otherwise I'll see if I fix it tomorrow.

joaquimorg, 2021-03-02:
Same problem with 
**Apache Newt 1.9.0-dev / 5da6e2f / 2021-03-02_11:04**

**$ newt new myproj**
Downloading project skeleton from apache/mynewt-blinky...
Downloading repository mynewt-blinky (commit: master) from https://github.com/apache/mynewt-blinky.git
Installing skeleton in myproj (commit: master)
Project myproj successfully created.

**$ newt upgrade**
Downloading repository mynewt-core (commit: master) from https://github.com/apache/mynewt-core.git
Downloading repository mynewt-nimble (commit: master) from https://github.com/apache/mynewt-nimble.git
Downloading repository mcuboot (commit: master) from https://github.com/JuulLabs-OSS/mcuboot.git
Downloading repository mynewt-mcumgr (commit: master) from https://github.com/apache/mynewt-mcumgr.git
Error: some repos are in a dirty state:
    apache-mynewt-core: contains local changes
Specify the `-f` (force) switch to attempt anyway

**$ newt upgrade -f**
WARNING: some repos are in a dirty state:
    apache-mynewt-core: contains local changes

Making the following changes to the project:
    upgrade apache-mynewt-core (0.0.1 --> 1.8.0)
    upgrade apache-mynewt-mcumgr (0.0.0 --> 0.1.0)
    upgrade apache-mynewt-nimble (0.0.0 --> 1.3.0)
    upgrade mcuboot (0.0.0 --> 1.5.0)
Error: Error updating "apache-mynewt-core": error: Your local changes to the following files would be overwritten by checkout:
        .mailmap
        .rat-excludes
        .style_ignored_dirs
        .travis.yml
        LICENSE
        apps/bleprph_oic/pkg.yml
  ......

utzig, 2021-03-02:
@joaquimorg The original issue is known, and it's because of submodules being moved around in the mcuboot repo, but yours looks more like a network issue or something similar, at least I don't know of any such issues in `mynewt-core`. Could you try cleaning up the repo, `pushd repos/apache-mynewt-core && git reset --hard && popd`, and running `newt upgrade` again? I believe the `-f` should do something like that already but have to review it again...

joaquimorg, 2021-03-02:
Thanks for the replay, but the error is the same:

**$ pushd repos/apache-mynewt-core && git reset --hard && popd**
/mnt/d/Work/mynewt_demos/myproj/repos/apache-mynewt-core /mnt/d/Work/mynewt_demos/myproj
Updating files: 100% (6307/6307), done.
HEAD is now at b4dae179a Merge pull request #2512 from utzig/kinetis-updates
/mnt/d/Work/mynewt_demos/myproj

**$ newt upgrade**
Error: some repos are in a dirty state:
    apache-mynewt-core: contains local changes
Specify the `-f` (force) switch to attempt anyway

utzig, 2021-03-02:
Hmmm, what OS are you using, is the filesystem mounted under `/mnt/d` (or `/mnt/d/Work` whatever) case-sensitive?

joaquimorg, 2021-03-02:
Hi, I am using Ubuntu in windows WSL, and yes is case-sensitive.

joaquimorg, 2021-03-02:
Even with the error I managed to compile the example.

sjanc, 2021-03-31:
1.9.0-rc1 is out, this issues should be fixed there
https://dist.apache.org/repos/dist/dev/mynewt/apache-mynewt-1.9.0/rc1/

sjanc, 2021-04-08:
I've tested this on linux and windows with 1.9.0 and all seem to work just fine. Is this stil an issue?

closed 2021-04-27 by sjanc