apache/mynewt-core #2535: Compiler packages for appcore and netcore of nRF5340

simonratner, 2021-03-17:
Commit disabling DSP instructions for M-33 suggests creating a custom compiler package for BSPs that need it: https://github.com/apache/mynewt-core/commit/bafc7e0e08057502d26fd7b84c89ac341c80e970.

Given that nRF5340 appcore and netcore BSPs are in-tree, and appcore supports DSP instruction set while the netcore does not, I think the two compiler packages themselves should be in the tree: `arm-none-eabi-m33` and `arm-none-eabi-m33-nodsp`, so that the corresponding Nordic BSPs can be used directly from the repo.

vrahane, 2021-04-14:
So, this can be solved by specifying `pkg.cflags:` for the respective BSP. apps, BSPs, targets and compiler packages when provided with `pkg.cflags` get applied to the entire build. cc: @ccollins476ad @vikrant-proxy

vrahane, 2021-04-14:
So, I think we can resolve this issue.

simonratner, 2021-04-15:
Yes, although I am not sure what the result of specifying multiple `-mcpu` options is. Would this prevent the mcu package from specifying a default? 

I have not tested this with cflags, but with lflags the order of options is also not preserved currently, which could make this trickier (https://github.com/apache/mynewt-newt/issues/429).